GyverLibs/EncButton #2: Файл FastIO.h не защищен от повторного включения

khusamov, 2021-07-17:
Здравствуйте!
Пытаюсь включить библиотеку и получаю такую ошибку:

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (3.1.0) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EncButton> 1.6.1
|-- <TVout> 1.0.1
Building in release mode
Compiling .pio\build\uno\src\Gamepad.cpp.o
Compiling .pio\build\uno\src\main.cpp.o
Linking .pio\build\uno\firmware.elf
.pio\build\uno\src\main.cpp.o: In function `fastRead(unsigned char)':
main.cpp:(.text._Z8fastReadh+0x0): multiple definition of `fastRead(unsigned char)'
.pio\build\uno\src\Gamepad.cpp.o:Gamepad.cpp:(.text._Z8fastReadh+0x0): first defined here
.pio\build\uno\src\main.cpp.o: In function `fastWrite(unsigned char, bool)':
main.cpp:(.text._Z9fastWritehb+0x0): multiple definition of `fastWrite(unsigned char, bool)'
.pio\build\uno\src\Gamepad.cpp.o:Gamepad.cpp:(.text._Z9fastWritehb+0x0): first defined here
.pio\build\uno\src\main.cpp.o: In function `fastShiftIn(unsigned char, unsigned char, unsigned char)':
main.cpp:(.text._Z11fastShiftInhhh+0x0): multiple definition of `fastShiftIn(unsigned char, unsigned char, unsigned char)'
.pio\build\uno\src\Gamepad.cpp.o:Gamepad.cpp:(.text._Z11fastShiftInhhh+0x0): first defined here
.pio\build\uno\src\main.cpp.o: In function `fastShiftOut(unsigned char, unsigned char, unsigned char, unsigned char)':
main.cpp:(.text._Z12fastShiftOuthhhh+0x0): multiple definition of `fastShiftOut(unsigned char, unsigned char, unsigned char, unsigned char)'
.pio\build\uno\src\Gamepad.cpp.o:Gamepad.cpp:(.text._Z12fastShiftOuthhhh+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\uno\firmware.elf] Error 1
```

Возникает она на таких трех файлах:

### main.cpp

```
#include <Arduino.h>
#include <Gamepad.h>

void setup() {}
void loop() {}
```

### Gamepad.cpp

```
#include "Gamepad.h"
```

### Gamepad.h

```
#pragma once

#include <EncButton.h>
```

GyverLibs, 2021-07-17:
Привет, защищён
'''#ifndef FastIO_h
#define FastIO_h'''

khusamov, 2021-07-17:
А почему у меня ошибка эта выходит?

GyverLibs, 2021-07-17:
Названия функций совпадают между библиотеками

khusamov, 2021-07-17:
Эти функции совпадают? fastRead fastWrite fastShiftIn fastShiftOut

Ваша библиотека конфликтует с библиотекой Arduino.h ?

GyverLibs, 2021-07-17:
Нет конечно, в Arduino.h нет таких функций, это ведь библиотека для ардуино

khusamov, 2021-07-17:
В таком случае не с чем совпасть. У меня в проекте всего три файла.

С чем тогда совпало? И как это исправить?

GyverLibs, 2021-07-17:
Написано что эти функции first defined in gamepad.cpp. Нужно переписывать либу. Могу выкатить апдейты на все либы с участием fastio, но оберну всё в namespace

khusamov, 2021-07-17:
Не понял. То есть ваша библиотека конфиктует сама с собой???

GyverLibs, 2021-07-17:
Gamepad.cpp не моя библиотека

khusamov, 2021-07-17:
Gamepad.cpp это мой файл. Но в нем нет функций с такими именами:

```cpp
#include "Gamepad.h"

int Gamepad::multTo5(int value) {
	return value * 5;
}

void Gamepad::multTo7(int* value) {
	*value = *value * 7;
}
```

GyverLibs, 2021-07-17:
Ну тогда все вопросы к платформио, я хз что он не может собрать

khusamov, 2021-07-17:
> Могу выкатить апдейты на все либы с участием fastio, но оберну всё в namespace

Выкатите, пожалуйста! Если это решит проблему.

khusamov, 2021-07-17:
> Ну тогда все вопросы к платформио, я хз что он не может собрать

О как. Я сейчас попробую в Arduino IDE воспроизвести эту ошибку. Любопытно стало...

khusamov, 2021-07-17:
[image]

GyverLibs, 2021-07-17:
А блин, encbutton подключен в другую библу и залетает туда в класс? Гляну, я в таком режиме не тестировал)

khusamov, 2021-07-17:
Спасибо! Жду решения) Я уже смотрел файлы ваши, но не допер в чем проблема.

khusamov, 2021-07-17:
[src.zip](https://github.com/GyverLibs/EncButton/files/6835125/src.zip)
Прикрепляю файлы для тестирования.

GyverLibs, 2021-07-17:
разобрался, пришлось чуть переделать fastIO. Сейчас выкачу обновы ко всем либам где он используется, спасибо за репорт

khusamov, 2021-07-18:
Спасибо! Компилируется без ошибок. Даже флаг

```
build_unflags = -flto
```

убрал. Все равно компилируется. Ранее без этого флага тоже были какие-то ошибки.

closed 2021-07-18 by khusamov

GyverLibs, 2021-07-18:
Я там заодно ещё оптимизации завёз, крупный скетч с разными режимами библиотеки весит на 1 КБ меньше)