GyverLibs/EncButton #4: Можно подключать кнопки из I2C-расширителя портов?

khusamov, 2021-07-28:
Думал виртуальные кнопки как раз для этого. Ан-нет. Не совсем понятно как такие кнопки подключать к вашей библиотеке.

AlexGyver, 2021-07-28:
Привет, как раз для этого и предусмотрен виртуальный режим. Опрашивать расширитель любыми своими способами и передавать значение в тик, как в примере

khusamov, 2021-07-28:
Попробовал. Но похоже на ESP8266 именно виртуальные кнопки не работают. 

То есть библиотека на ESP8266 нормально работает. Но я попробовал виртуальные кнопки и ничего не вышло. На вызове tick() происходит перезагрузка МК.

khusamov, 2021-07-28:
```cpp
#include <EncButton.h>
#include <PCF8574.h>
#include <Wire.h>

EncButton<EB_TICK, VIRT_BTN> enc;     // виртуальная кнопка
PCF8574 pcf20(0x20);

void setup() {
  pcf20.begin();
}

void loop() {
  bool state = this->pcf20->readButton(0);
  enc.tick(!state );
}
```

GyverLibs, 2021-07-28:
Всё замечательно работает на есп. Приведённый выше скетч даже не скомпилится, смотри примеры работы с PCF8574 в папке examples

khusamov, 2021-07-28:
Я делал на основе https://github.com/RobTillaart/PCF8574/blob/master/examples/buttonRead/buttonRead.ino
куда еще проще. Вроде ничего не пропустил...

GyverLibs, 2021-07-28:
повторюсь, у меня работает отлично)

GyverLibs, 2021-07-28:
[image]

khusamov, 2021-07-29:
Приношу извинения, да, действительно, пример я привел странный. 

Я подготовил точный пример из моего кода. Убрал все лишнее, оставил только расширитель, одну кнопку и EncButton. Симптомы проблемы правда другие. При старте в мониторе порта появляется следующее:

```
Start
Gamepad.isButtonPress(0)
```

Нажатия на кнопку ни к чему не приводят, хотя при нажатии должна появляться строка `Gamepad.isButtonPress(0)`.

Вот код:

**Файл `main.cpp`:**

```cpp
#include "Gamepad.h"

Gamepad gamepad;

void setup() {
	Serial.begin(9600);
	Serial.println();
	Serial.println("Start");
	gamepad.begin();
}

void loop() {
	gamepad.tick();
	if (gamepad.isButtonPress()) {
		Serial.println("Gamepad.isButtonPress(0)");
	}
}
```

**Файл `Gamepad.h`:**

```cpp
#pragma once
#include <Arduino.h>
#include <PCF8574.h> // https://github.com/RobTillaart/PCF8574
#include <EncButton.h>

class Gamepad {
	public:
		Gamepad();
		void begin();
		void tick();
		bool isButtonPress();
	private:
		PCF8574 *pcf20 = nullptr;
		EncButton<EB_TICK, VIRT_BTN> *button = nullptr;
};
```

**Файл `Gamepad.cpp`:**

```cpp
#include <Arduino.h>
#include "Gamepad.h"

Gamepad::Gamepad() {
	this->pcf20 = new PCF8574(0x20);
	this->button = new EncButton<EB_TICK, VIRT_BTN>;
}

void Gamepad::begin() {
	this->pcf20->begin();
}

void Gamepad::tick() {
	this->button->tick(!this->pcf20->readButton(0));
}

bool Gamepad::isButtonPress() {
	return this->button->isPress();
}
```

GyverLibs, 2021-07-29:
Ох, дичь какая. Не знаю насколько корректно будет работать динамическое создание шаблонного класса, не пробовал. Но по коду -  `readButton()` вызывается без указания пина, что судя по PCF8574.h не должно компилироваться

khusamov, 2021-07-29:
Нет, этот код я отладил и залил в МК. Нулик это я пропустил тут, когда копировал в пост.

khusamov, 2021-07-29:
Ну дичь не дичь, но в реальных проектах будут классы. И создаваться объекты библиотеки будут по разному. Это уже не тестовый пример все-таки, это попытка написать более-менее полноценную игру. Я даже плату спаял для этого.

GyverLibs, 2021-07-29:
Нда... Попробуй тогда сначала статически создать encbutton, если заработает - уже переходить на динаму и тысячу и один this->. Я ООП головного мозга не страдаю, возможно поэтому есть какие то недоработки в либе, ничего не могу сказать на этот счёт

khusamov, 2021-07-29:
Попробовал статически. Работает. Правда не стабильно. В конце связь обрывается.

```
Start
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Gamepad.isButtonPress
Exception in thread rx:
Traceback (most recent call last):
  File "C:\Users\khusamov\.platformio\python3\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\khusamov\.platformio\python3\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\khusamov\.platformio\penv\lib\site-packages\serial\tools\miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "c:\users\khusamov\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 295, in read
    raise SerialException("GetOverlappedResult failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: GetOverlappedResult failed (PermissionError(13, 'Отказано в доступе.', None, 5))
```

**Файл `main.cpp`:**

```cpp
#include "Gamepad.h"

Gamepad gamepad;

void setup() {
	Serial.begin(9600);
	Serial.println();
	Serial.println("Start");
	gamepad.begin();
}

void loop() {
	gamepad.tick();
	if (gamepad.isButtonPress()) {
		Serial.println("Gamepad.isButtonPress");
	}
}
```

**Файл `Gamepad.h`:**

```cpp
#pragma once
#include <Arduino.h>
#include <PCF8574.h> // https://github.com/RobTillaart/PCF8574
#include <EncButton.h>

class Gamepad {
	public:
		Gamepad();
		void begin();
		void tick();
		bool isButtonPress();
	private:
		PCF8574 *pcf20 = nullptr;
		EncButton<EB_TICK, VIRT_BTN> button;
};

```

**Файл `Gamepad.cpp`:**

```cpp
#include <Arduino.h>
#include "Gamepad.h"

Gamepad::Gamepad() {
	this->pcf20 = new PCF8574(0x20);
}

void Gamepad::begin() {
	this->pcf20->begin();
}

void Gamepad::tick() {
	button.tick(!this->pcf20->readButton(0));
}

bool Gamepad::isButtonPress() {
	return button.isPress();
}
```

GyverLibs, 2021-07-29:
ошибка не выглядит как проблема с encbutton

khusamov, 2021-07-29:
Какая? С обрывом связи? Да бог с ней. Главное - что делать с динамическим созданием объектов твоей библиотеки?

GyverLibs, 2021-07-29:
> Правда не стабильно. В конце связь обрывается.

Здесь тоже на encbutton подозрение? Или я не понял к чему вообще

khusamov, 2021-07-29:
> > Правда не стабильно. В конце связь обрывается.
> 
> Здесь тоже на encbutton подозрение? Или я не понял к чему вообще

Пока нет подозрений. Просто она есть и она повторяется. Главное что нажатия фиксируются. 

В общем я понял, динамическое создание объектов твоя библиотека не выдерживает. Пойду думать что с этим делать. Спасибо за отклик!

GyverLibs, 2021-07-29:
Что значит "твоя библиотека не выдерживает"? Может питон не в состоянии правильно собрать прошивку или программист где то допускает ошибки? 
**Скетч**
```
#define BTN_PIN 3
#include "lib.h"
Test test;

void setup() {
  Serial.begin(9600);
  pinMode(BTN_PIN, INPUT_PULLUP);
}

void loop() {
  test.tick();
  if (test.click()) Serial.println("click");  
}
```
**lib.h**
```
#include <EncButton.h>
class Test {
  public:
    Test() {
      btn = new EncButton<EB_TICK, VIRT_BTN>;
    }
    void tick() {
      btn->tick(digitalRead(BTN_PIN));
    }
    bool click() {
      return btn->isClick();
    }

    EncButton<EB_TICK, VIRT_BTN> *btn;
  private:
};
```
[image]

khusamov, 2021-07-29:
Спасибо! Все заработало!
В итоге скетч не менял свой, просто на 1001-й раз он просто заработал. 

Правда при последущих разах загрузки скетча в МК, то работает, то не работает.

GyverLibs, 2021-07-29:
Программирование это религия, чем больше бьешься лбом об пол и молишься, тем больше шанс что тот же код заработает))

khusamov, 2021-07-29:
У меня есть подозрения, что я спаял плохо. Первая плата все-таки. USB например то пашет, то не пашет. 
В общем спасибо! Главное я убедился, что со софтом все в порядке.

GyverLibs, 2021-07-29:
Ура ура

closed 2021-07-29 by GyverLibs