GyverLibs/EncButton #5: Как задать #define EB_HOLD для разных кнопок разные значения?

khusamov, 2021-07-29:
Добрый день!

Параметр #define EB_HOLD влияет на все кнопки. А как задать этот параметр для кнопок отдельно?

Например для вращения пушки нужно значение нуль, а вот для кнопки выстрела очередями нужна небольшая задержка.

GyverLibs, 2021-07-29:
Аааа к сожалению никак, библиотека создавалась как максимально оптимальная для мк без байта лишнего кода, всё на константах. Нужный функционал можно добавить у себя локально, там пару строчек дописать

khusamov, 2021-07-29:
То есть не для игр библиотека. Понимаю.

А локально это как? Скопировать к себе и править под себя? Или можно как-то без копирования сделать?

GyverLibs, 2021-07-29:
Ну, скачать себе библу и добавить переменную вместо константы

GyverLibs, 2021-07-31:
Сделал индивидуальную настройку удержания в 1.8

closed 2021-09-09 by GyverLibs