GyverLibs/EncButton #6: Прерывания и pcf8575

pruwait, 2021-08-15:
Прошу добавить пример для отлова даблкликов и холдов с использованием прерываний на esp32. 
Пока не очень понятно, когда и сколько раз вызывать tick():
в лууп, в функции прерывания, в обоих случаях.

GyverLibs, 2021-08-15:
И там и там, потому что клик отлавливается по тайм-ауту, который в прерывании уже не тикнется

pruwait, 2021-08-15:
Есть какая-то компактная конструкция для объявления 32х виртуальных кнопок?
Такое сработает?
EncButton * buttons = new EncButton<EB_TICK, VIRT_BTN> [32]
или
EncButton<EB_TICK, VIRT_BTN> * buttons = new EncButton [32]

GyverLibs, 2021-08-15:
Конечно

pruwait, 2021-08-15:
Соединения esp32 и pcf8575

   PCF8575    ----- Esp32
   A0         ----- GRD
   A1         ----- GRD
   A2         ----- GRD
   VSS        ----- GRD
   VDD        ----- 5V/3.3V
   SDA        ----- 21
   SCL        ----- 22
   INT        ----- 32,33   

   P0     ----------------- SW0
   P1     ----------------- SW1
   P2     ----------------- SW2
   P3     ----------------- SW3
   P4     ----------------- SW4
   P5     ----------------- SW5
   P6     ----------------- SW6
........................................................
   P15     ----------------- SW15

Фрагмент кода:
////////////////////////////////////
#include "PCF8575.h"  //https://github.com/xreef/PCF8575_library 
//для использования раскомметнировать в *.h // #define PCF8575_LOW_MEMORY

#include <ETH.h>
#include "lib/PubSubClient/src/PubSubClient.h"

//Все GPIO могут быть настроены для обработки прерываний. По прерываниям обновляются статусы виртуальных входов
#define ESP32_INTERRUPTED_PIN1 32
#define ESP32_INTERRUPTED_PIN2 33

//адреса двух расширителей для 32х входов
#define addrIn1 0x20
#define addrIn2 0x21

#include <EncButton.h> // для отслеживания кликов, даблкликов, лонгпрессов

EncButton<EB_TICK, VIRT_BTN> * buttons = new EncButton<EB_TICK, VIRT_BTN> [32]; // массив виртуальных кнопок

// Устанавливаем адреса i2c расширителей, пины и функции прерываний 
PCF8575 in1(0x21, ESP32_INTERRUPTED_PIN1, keyChangedOnPCF1);
PCF8575 in2(addrIn2, ESP32_INTERRUPTED_PIN2, keyChangedOnPCF2);

void setup() {
 // Устанавливаем pinMode расширителей
  for (uint8_t i = 0; i < 16; i++) {
   in1.pinMode(i, INPUT);
    in2.pinMode(i, INPUT);
  }

// тут под вопросом к разработчику библиотеки расширителей, возможно обойдемся без бегинов
  in1.begin(); 
  in2.begin();

  timeElapsed = millis(); // для дальнейших периодических проверок в луп
} // setup

void loop() {
  if (keyChanged1) { //при изменении входов первого расширителя
    byte di = in1.digitalReadAll(); // здесь состояния всех входов после изменения как 11100011110001

    //разбираем байты и тикаем виртуальные кнопки в цикле
    /*
       p0 = ((di & bit(0))>0)?HIGH:LOW;
       p1 = ((di & bit(1))>0)?HIGH:LOW;
       p2 = ((di & bit(2))>0)?HIGH:LOW;
       p3 = ((di & bit(3))>0)?HIGH:LOW;
       p4 = ((di & bit(4))>0)?HIGH:LOW;
       p5 = ((di & bit(5))>0)?HIGH:LOW;
       p6 = ((di & bit(6))>0)?HIGH:LOW;
       p7 = ((di & bit(7))>0)?HIGH:LOW;
    */
       // возможно, этот цикл придется повторить в функции обработки прерывания?
       for (uint8_t i = 0; i < 16; i++) { // тикаем первые 16 виртуальных кнопок, т.е. buttons[0-15]
        buttons[i].tick(((di & bit(i))>0)?HIGH:LOW); // здесь под вопросом, возможно придется инвертировать вроде ((di & bit(7))>0)?LOW:HIGH;
  } //for

  keyChanged1 = false;  //сбрасываем флаг изменённых входов расширителя
} //if

if (keyChanged2) { //при изменении входов второго расширителя
  byte di = in2.digitalReadAll(); // здесь состояния всех входов после изменения как 11100011110001

    //разбираем байты и тикаем виртуальные кнопки в цикле
    /*
       p0 = ((di & bit(0))>0)?HIGH:LOW;
       p1 = ((di & bit(1))>0)?HIGH:LOW;
       p2 = ((di & bit(2))>0)?HIGH:LOW;
       p3 = ((di & bit(3))>0)?HIGH:LOW;
       p4 = ((di & bit(4))>0)?HIGH:LOW;
       p5 = ((di & bit(5))>0)?HIGH:LOW;
       p6 = ((di & bit(6))>0)?HIGH:LOW;
       p7 = ((di & bit(7))>0)?HIGH:LOW;
    */
       // возможно, этот цикл придется повторить в функции обработки прерывания?
       for (uint8_t i = 0; i < 16; i++) { // тикаем вторые 16 виртуальных кнопок, т.е. buttons[16-31]
        buttons[i+16].tick(((di & bit(i))>0)?HIGH:LOW); // здесь под вопросом, возможно придется инвертировать вроде ((di & bit(7))>0)?LOW:HIGH;
  } //for

  keyChanged2 = false;  //сбрасываем флаг изменённых входов расширителя

} //if

} // loop

// Функции прерываний

void  keyChangedOnPCF1() {
  // Interrupt called (No Serial no read no wire in this function, and DEBUG disabled on PCF library)
  keyChanged1 = true;

// Здесь нужно засунуть цикл считывания входов расширителя с дальнейшим шестнадцатикратным тиканием виртуальных кнопок?

}

void  keyChangedOnPCF2() {
  // Interrupt called (No Serial no read no wire in this function, and DEBUG disabled on PCF library)
  keyChanged2 = true;

// Здесь нужно засунуть цикл считывания входов расширителя с дальнейшим шестнадцатикратным тиканием виртуальных кнопок?
}
//////////////////////////////////////////// конец кода

Код компилируется, но реальной проверки не было. Жду расширители через неделю.

Пока вопрос по месту тика остался.
Обращаю внимание на прерывания в ESP32 (возможно они какие-нибудь софтовые и работают иначе).
Обращаю внимание на предупреждения от разработчика библиотеки расширителя: No Serial no read no wire in this function, and DEBUG disabled on PCF library

Если засунуть в функции обработчика прерываний считывание входов расширителя + разбор значения на байты + тикание виртуальных кнопок - всё ли отработает корректно?
На esp32 планируется еще ethernet поднять с рапортованием по mqtt на сервер умного дома... Не убьют ли сеть эти чрезмерно жирные функции обработчиков прерываний?

GyverLibs, 2021-08-15:
Вызывать в прерывании и лупе. Жира у есп32 хватит на всё, если что можно на второе ядро повесить

closed 2021-09-09 by GyverLibs

pruwait, 2021-09-12:
С прерываниями и расширителями портов на всё так просто.

GyverLibs, 2021-09-12:
значит почаще опрашивать расширитель, библиотека так и работает

pruwait, 2021-09-12:
Если просто очень часто опрашивать расширитель, теряется смысл прерываний и подключения пина INT от pcf8575.
Идея была фиксировать нажатие только тогда, когда оно происходит, а не всё время работы контроллера.

Если взвожу в прерывании флаг keyChanged1 = true и такаю только по нему в лупе - фиксируются только клики. 
Как же сделать даблклики и холды с прерываниями?
Неужели нет штатного алгоритма для этого случая?

GyverLibs, 2021-09-12:
очевидно, никак. Холды, клики и дебаунсы основаны на времени. Если у тебя тикер только по состоянию кнопки - дебаунс в прерывании отловить не получится. Где-то ещё должен опрашиваться таймер и решить, что прошло нужное время. Хосподи, в есп32 достаточно жира для любых задач, просто вызывай тикер дополнительно хотя бы каждые 10 мс, этого достаточно. Там ведь даже RTOS есть, ну ёлки палки =)

pruwait, 2021-09-12:
Вроде заработало. 
Без нажатия не кнопки лишний раз не тикает.
По нажатию на любую кнопку вызывается прерывание с взведением флага нажатия и запоминанием времени нажатия.

В лупе тикаю кнопки сразу после прерывания и в течение 5 секунд далее с интервалом 10 мс.
Холд более 5 секунд в моём проекте не будет использоваться.

```
void loop_buttons() {
  
   if (millis() - timing > 10){ // Проверяем кнопки каждые 10мс
  timing = millis(); 
  
   
  //обработка выключателей
  if ((keyChanged1) || (millis() - time_click1 < 5000)) { //при изменении входов первого расширителя или в течение 5 секунд после
    uint16_t di = in1.readButton16(); // здесь состояния всех входов после изменения как 11100011110001, в прерывании эта функция вызывает перезагрузку
    //разбираем байты и тикаем виртуальные кнопки в цикле
    /*
       p0 = ((di & bit(0))>0)?HIGH:LOW;
       p1 = ((di & bit(1))>0)?HIGH:LOW;
       p2 = ((di & bit(2))>0)?HIGH:LOW;
       p3 = ((di & bit(3))>0)?HIGH:LOW;
       p4 = ((di & bit(4))>0)?HIGH:LOW;
       p5 = ((di & bit(5))>0)?HIGH:LOW;
       p6 = ((di & bit(6))>0)?HIGH:LOW;
       p7 = ((di & bit(7))>0)?HIGH:LOW;
    */
    for (uint8_t i = 0; i < 16; i++) { // тикаем первые 16 виртуальных кнопок, т.е. buttons[0-15]
      buttons[i].tick(((di & bit(i)) > 0) ? LOW : HIGH); // кнопки подтянуты к питанию и прижимаем к GND в нажатии, поэтому HIGH = отпущена

      if (buttons[i].isClick()) Serial.println("click");
      if (buttons[i].isHolded()) Serial.println("holded");
      // if (buttons[i].isStep()) Serial.println("step");

      if (buttons[i].isPress()) Serial.println("press");
      if (buttons[i].isClick()) Serial.println("click");
      // if (buttons[i].isRelease()) Serial.println("release");

      if (buttons[i].hasClicks(1)) Serial.println("1 click");
      if (buttons[i].hasClicks(2)) Serial.println("2 click");
      if (buttons[i].hasClicks(3)) Serial.println("3 click");
      if (buttons[i].hasClicks(5)) Serial.println("5 click");

      if (buttons[i].hasClicks()) Serial.println(buttons[i].clicks);
      if (buttons[i].isHold()) Serial.println("hold");

    } //for

    keyChanged1 = false; //сбрасываем флаг изменённых входов расширителя
}//if  флаг прерывания

 } //if timing
} // loop
```

GyverLibs, 2021-09-12:
порнуха, но ладно)