GyverLibs/EncButton #8: Guru Meditation Error на ESP32

bill1389, 2021-12-26:
Решил использовать библиотеку в проекте бортового компьютера на esp32. Если использовать тикер в loop, то все нормально, если пользоваться прерываниями, то периодически при нажатии кнопки все зависает с ошибкой в консоли: Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400d3308  PS      : 0x00060034  A0      : 0x40083958  A1      : 0x3ffbec10  
A2      : 0x00000011  A3      : 0x3ffc02cc  A4      : 0x00000000  A5      : 0x00020000  
A6      : 0x3ffae270  A7      : 0x00060323  A8      : 0x80080f08  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ffb932c  A12     : 0x80082cde  A13     : 0x3ffb1a50  
A14     : 0x00000001  A15     : 0x3ffb8058  SAR     : 0x0000001b  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4008a9c2  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d3308

ELF file SHA256: 0000000000000000

Backtrace: 0x400d3308:0x3ffbec10 0x40083955:0x3ffbec30 0x4008a9bf:0x3ffb1a80 0x4008aa86:0x3ffb1ab0 0x4008b2d5:0x3ffb1ad0 0x4008

GyverLibs, 2021-12-26:
Ох уж эта есп...

bill1389, 2021-12-26:
Без шансов на ней использовать библиотеку?

GyverLibs, 2021-12-26:
Я не работал с прерываниями на есп и не знаю, какие там есть подводные камни. Скорее всего без проблем можно завести, но мне нужно больше информации. 

Так, какой тикер используется в прерывании? Обычный или isr?

bill1389, 2021-12-26:
btnUp.tickISR();

GyverLibs, 2021-12-26:
А если просто тик

bill1389, 2021-12-26:
Пробовал, тоже самое...Да ладно, что-нибудь придумаю, мне по сути нужно 2 действия кнопки, клик и удержание. Я думал вдруг проблема известная, и есть готовое решение. Спасибо.

GyverLibs, 2021-12-27:
Можно попробовать отладить, у меня нет есп32 на руках. Закомментируй строчку `private:` в библиотеке. И в прерывании пиши вместо тик делаем
```
_btnState = digitalRead пин
poolBtn();
```
Надо локализовать проблему

bill1389, 2021-12-27:
Уточню, в библиотеке, я комментирую только 282 строчку private:
в скетче делаю так?
void isrUp() {
  btnUp._btnState = digitalRead(KEY_UP);
  btnUp.poolBtn();  // опрашиваем в прерывании, чтобы поймать нажатие в любом случае
}

в setup: attachInterrupt(KEY_UP, isrUp, CHANGE);

Что-то так не работают нажатия. Я в принципе уже вышел из положения, запустил тикер на втором ядре в цикле)

GyverLibs, 2021-12-27:
Не так, под esp прерывания объявляются по другому

bill1389, 2021-12-27:
если имеется ввиду attachInterrupt(digitalPinToInterrupt(KEY_UP), isrUp, CHANGE); - то так тоже пробовал, никакой разницы.

GyverLibs, 2021-12-27:
Нет, обработчик объявляется с атрибутом IRAM

closed 2022-01-08 by GyverLibs