GyverLibs/EncButton #9: Много warning при компиляции

demosspro, 2022-01-01:
Подскажите - это нормально - такое количество ворнингов при компиляции?
1.8.16 софт, компилятор (5.4 или 7.3)

[image]

GyverLibs, 2022-01-04:
Как получить столько варнингов? У меня ни одного

demosspro, 2022-01-05:
`// TX
// подключаем библу
#define STRIP_PIN 6     // пин ленты
#define NUMLEDS 1      // кол-во светодиодов
#define COLOR_DEBTH 2
#include <microLED.h>
microLED<NUMLEDS, STRIP_PIN, MLED_NO_CLOCK, LED_WS2812, ORDER_GRB, CLI_AVER> strip;

#define EB_CLICK 600
#include <EncButton.h>
#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"
#include <GyverPower.h>

volatile boolean isrflag = true;
//void(* resetFunc) (void) = 0;
//void isr() {
//resetFunc();
//}
EncButton<EB_TICK, 2> btn0;
EncButton<EB_TICK, 3> btn1;
EncButton<EB_TICK, 4> btn2;
EncButton<EB_TICK, 5> btn3;

RF24 radio(9, 10); // "создать" модуль на пинах 9 и 10 Для Уно
byte address[][6] = {"1Node", "2Node", "3Node", "4Node", "5Node", "6Node"}; //возможные номера труб
uint16_t transmit_data[1]; // массив, хранящий передаваемые данные
uint16_t latest_data[1]; // массив, хранящий последние переданные данные
boolean flag; // флажок отправки данных
uint32_t timer0, timer1;
double voltage_r = 3.49;
double voltage_t = 3.49;
int Pstate = 0;
int  pled = 7;

void setup() {
  //Serial.begin(56700);
  analogReference(INTERNAL);
  pinMode(A0, INPUT);
  pinMode(pled, OUTPUT);
  digitalWrite(pled, LOW);

  //Serial.println("led_power");
  strip.setBrightness(20);

  radio.begin();
  radio.setAutoAck(1);         //режим подтверждения приёма, 1 вкл 0 выкл
  radio.setRetries(0, 15);    //(время между попыткой достучаться, число попыток)
  radio.enableAckPayload();    //разрешить отсылку данных в ответ на входящий сигнал
  radio.setPayloadSize(32);     //размер пакета, в байтах

  radio.openWritingPipe(address[0]);   //мы - труба 0, открываем канал для передачи данных
  radio.setChannel(0x66);  //выбираем канал (в котором нет шумов!)

  radio.setPALevel (RF24_PA_MAX); //уровень мощности передатчика. На выбор RF24_PA_MIN, RF24_PA_LOW, RF24_PA_HIGH, RF24_PA_MAX
  radio.setDataRate (RF24_1MBPS); //скорость обмена. На выбор RF24_2MBPS, RF24_1MBPS, RF24_250KBPS
  //должна быть одинакова на приёмнике и передатчике!
  //при самой низкой скорости имеем самую высокую чувствительность и дальность!!

  radio.powerUp(); //начать работу
  radio.stopListening();  //не слушаем радиоэфир, мы передатчик
  power.autoCalibrate(); // автоматическая калибровка ~ 2 секунды , средняя но достаточная точность
  // см раздел константы в GyverPower.h, разделяющий знак " | "
  power.setSleepMode(POWERDOWN_SLEEP); // если нужен другой режим сна, см константы в GyverPower.h (по умолчанию POWERDOWN_SLEEP)
  power.bodInSleep(false); // рекомендуется выключить bod во сне для сохранения энергии (по умолчанию false - выключен!!)
  //attachInterrupt(digitalPinToInterrupt(2), wkp, FALLING);

}
[... 130 lines cut ...]
      latest_data[i] = transmit_data[i]; // запомнить последнее изменение
    }
  }

  if (flag == 1) {
    radio.powerUp(); // включить передатчик
    radio.write(&transmit_data, sizeof(transmit_data));       // отправить по радио
    //Serial.println(latest_data[0]);
    flag = 0; //опустить флаг
    if (radio.available() ) {                   // если в ответе что-то есть
      radio.read(&voltage_r, sizeof(voltage_r));    // читаем
      //Serial.println(voltage_r);
      timer0 = millis();
      radio.powerDown(); // выключить передатчик
    }

  }
}
void wkp() {
  power.hardwareEnable(PWR_ADC | PWR_TIMER1);
  digitalWrite(pled, HIGH);
  power.autoCalibrate();
  isrflag = true;
  //Serial.print("WKPstart");
}`

Seriyv0lk, 2022-01-20:
> 

Аналогичная история. Варнинги один в один с ТС-ом. Так понимаю, решения нет?

GyverLibs, 2022-01-20:
Варнинг - не ошибка) если кто то поможет пофиксить непонятные варнинги от шаблона - их не будет

Seriyv0lk, 2022-01-20:
> Варнинг - не ошибка) если кто то поможет пофиксить непонятные варнинги от шаблона - их не будет

В общем, разобрался. В либе EncButton.h  в строке 384 нужно заменить это
`if (debounce < (_holdT << 7))`
на это
`if (debounce < (uint8_t)(_holdT << 7))`

Seriyv0lk, 2022-01-20:
Ну и спасибо тебе за труд, дружище)

GyverLibs, 2022-01-20:
Не, этот варнинг ерунда, с ним все понятно. Компилятору лень делать свою работу)

Seriyv0lk, 2022-01-20:
С ним и остальные пропадают.

GyverLibs, 2022-01-20:
`if (debounce < (uint8_t)(_holdT << 7))`
Не, так всё сломается, там ведь uint16_t вычисления

GyverLibs, 2022-01-20:
> С ним и остальные пропадают.

Ну и бред) тогда поправлю и релизну сейчас

Seriyv0lk, 2022-01-20:
> > С ним и остальные пропадают.
> 
> Ну и бред) тогда поправлю и релизну сейчас

Опечатался. Там сравнивается с 32-разрядным. Компилер его и хочет. 
А у меня было так до правки:

> `In file included from src\main.cpp:4:0:
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h: In instantiation of 'void EncButton<_EB_MODE, _S1, _S2, _KEY>::poolBtn() [with unsigned char _EB_MODE = 0; unsigned char _S1 = 1; unsigned char _S2 = 255; unsigned char _KEY = 255]':
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h:193:24:   required from 'uint8_t EncButton<_EB_MODE, _S1, _S2, _KEY>::tickISR(uint8_t, uint8_t, uint8_t) [with unsigned char _EB_MODE = 0; unsigned char _S1 = 1; unsigned char _S2 = 255; unsigned char _KEY = 255; uint8_t = unsigned char]'
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h:166:16:   required from 'uint8_t EncButton<_EB_MODE, _S1, _S2, _KEY>::tick(uint8_t, uint8_t, uint8_t) [with unsigned char _EB_MODE = 0; unsigned char _S1 = 1; unsigned char _S2 = 255; unsigned char _KEY = 255; uint8_t = unsigned char]'
> src\main.cpp:35:14:   required from here
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h:384:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
>                      if (debounce < ((uint8_t) _holdT << 7)) {                         // прошло меньше удержания
>                          ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h: In instantiation of 'void EncButton<_EB_MODE, _S1, _S2, _KEY>::poolBtn() [with unsigned char _EB_MODE = 0; unsigned char _S1 = 5; unsigned char _S2 = 255; unsigned char _KEY = 255]':
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h:193:24:   required from 'uint8_t EncButton<_EB_MODE, _S1, _S2, _KEY>::tickISR(uint8_t, uint8_t, uint8_t) [with unsigned char _EB_MODE = 0; unsigned char _S1 = 5; unsigned char _S2 = 255; unsigned char _KEY = 255; uint8_t = unsigned char]'
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h:166:16:   required from 'uint8_t EncButton<_EB_MODE, _S1, _S2, _KEY>::tick(uint8_t, uint8_t, uint8_t) [with unsigned char _EB_MODE = 0; unsigned char _S1 = 5; unsigned char _S2 = 255; unsigned char _KEY = 255; uint8_t = unsigned char]'
> src\main.cpp:36:13:   required from here
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h:384:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h: In instantiation of 'void EncButton<_EB_MODE, _S1, _S2, _KEY>::poolBtn() [with unsigned char _EB_MODE = 0; unsigned char _S1 = 4; unsigned char _S2 = 255; unsigned char _KEY = 255]':
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h:193:24:   required from 'uint8_t EncButton<_EB_MODE, _S1, _S2, _KEY>::tickISR(uint8_t, uint8_t, uint8_t) [with unsigned char _EB_MODE = 0; unsigned char _S1 = 4; unsigned char _S2 = 255; unsigned char _KEY = 255; uint8_t = unsigned char]'
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h:166:16:   required from 'uint8_t EncButton<_EB_MODE, _S1, _S2, _KEY>::tick(uint8_t, uint8_t, uint8_t) [with unsigned char _EB_MODE = 0; unsigned char _S1 = 4; unsigned char _S2 = 255; unsigned char _KEY = 255; uint8_t = unsigned char]'
> src\main.cpp:37:14:   required from here
> C:\Users\Sergey\Documents\Arduino\libraries\EncButton\src/EncButton.h:384:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]`

GyverLibs, 2022-01-20:
Самое смешное то в том, что _holdT - изначально unsigned. Непонятно как сдвиг заставляет компилятор решить что чисто стало signed

Seriyv0lk, 2022-01-20:
Меня тоже это сбило с толку сначала. Потом пошёл по цепочке и нашёл несовпадающие типы. Прописал и стало нормально компилиться. Что в ардуине, что в платформио.

GyverLibs, 2022-01-20:
В общем залил

closed 2022-01-20 by GyverLibs