GyverLibs/EncButton #10: Добавить поддержку энкодера

VSVLAD, 2022-01-02:
У меня есть энкодер, но прокрутка у него срабатывает только каждый второй раз. Насколько я понял, EncButton перестал поддерживать несколько типов энкодеров.

[image]

В локальной библиотечке у себя в методе:
void poolEnc(uint8_t state)

заменяю
`if (state == 0x3 && _ecount != 0) {`

на
`if ((state == 0x3 || state == 0x0) && _ecount != 0) {`

то всё работает как требуется.
Можете ещё один дефайн сделать и поправить код, чтобы поддержку этого энкодера включить?

bar2104y, 2022-01-09:
Поддерживаю, вариант решения выше работает

GyverLibs, 2022-01-09:
Замутил, спасибо за готовый кусок кода! =)

closed 2022-01-09 by GyverLibs