GyverLibs/EncButton #13: isActive

AdamKhagar, 2022-01-09:
хотелось бы иметь метод `isActive` который возвращал `bool`.

`isActive()` возвращает `true` если с энкодером/кнопкой происходит какое либо действие, в противном случае - `false`. Такой метод помог бы вынести всю логику связанную с обработкой действий кнопок в отдельную функцию

```
void loop() {
    enc.tick();
    if (enc.isActive()) {
         doSomething();
    }
    .....
}

void doSomething() {
    // Здесь вся логика связанная с обработкой действий с кнопкой
    .....
}

GyverLibs, 2022-01-09:
https://github.com/GyverLibs/EncButton/blob/main/examples/optimisation/optimisation.ino

AdamKhagar, 2022-01-09:
Спасибо, не заметил когда смотрел примеры

closed 2022-01-09 by AdamKhagar