GyverLibs/EncButton #14: Почему два класса

XRay3D, 2022-01-13:
В чём основное различие между EncButton и EncButton2
Возможно я могу помочь свести в один файл.

XRay3D, 2022-01-13:
В чём основное различие между EncButton и EncButton2?
Возможно я могу помочь свести в один файл.

GyverLibs, 2022-01-13:
Первый класс хранит номера пинов в шаблоне, второй - в переменных класса

XRay3D, 2022-01-13:
От этого большой выигрыш памяти скорости?
Могу попробовать написать обёртку для шаблонного варианта. Но не уверен что это можно сделать в с++11.

GyverLibs, 2022-01-13:
Выигрыш во всём - в шаблоне. Вторая версия класса сделана для удобного задания массива кнопок/энкодеров

XRay3D, 2022-01-13:
Ок тогда попробую завернуть EncButton.

XRay3D, 2022-01-16:
Похоже что виртуальными кнопками придётся пожертвовать или вынести отдельный класс.

GyverLibs, 2022-01-17:
да зачем, и так всё отлично организовано. Сейчас отдельный класс нужен только для массива физических кнопок

XRay3D, 2022-01-17:
> да зачем, и так всё отлично организовано. Сейчас отдельный класс нужен только для массива физических кнопок

Есть возможность всё в один свести.

XRay3D, 2022-01-24:
https://github.com/XRay3D/EncButton/tree/master
Борщенул с CRTP и выиграл от 50 до 102 байт по флешу.
Теперь всё в одном классе. 
Хотел ещё и недопустимые для выбранного типа проверки состояний вынести в ```ButtonImpl``` и ```EncoderImpl```

GyverLibs, 2022-01-24:
Божечки! При всем желании не смогу разобраться, как оно работает)

XRay3D, 2022-01-24:
> Божечки! При всем желании не смогу разобраться, как оно работает)

https://en.cppreference.com/w/cpp/language/partial_specialization
https://en.cppreference.com/w/cpp/language/crtp
https://en.cppreference.com/w/cpp/language/qualified_lookup
https://en.cppreference.com/w/cpp/language/adl

XRay3D, 2022-01-24:
https://en.cppreference.com/w/cpp/language/crtp
Это как "класс из кубиков лего собирать".

XRay3D, 2022-01-24:
Брать будете)) Дизайнил из-за обратной совместимости. Почти.

XRay3D, 2022-01-27:
https://youtu.be/-QOaUNFGj6E?t=2968

closed 2022-01-28 by XRay3D

XRay3D, 2022-02-10:
> Божечки! При всем желании не смогу разобраться, как оно работает)

Годнота по теме подъехала.
https://www.youtube.com/watch?v=UrL5gdW2JOM