GyverLibs/EncButton #19: Работа энкодера.

schpuntik, 2022-01-29:
У меня энкодер как   такой  же как здесь. https://github.com/GyverLibs/EncButton/issues/10   
Я прописал   дефайны вот так.
`#define EB_BETTER_ENC   // улучшенный алгоритм опроса энкодера. Добавит 16 байт SRAM при подключении библиотеки
#define EB_HALFSTEP_ENC // режим опроса полушагового энкодера (включи, если твой энкодер делает два тика за один)`
Но  энкодер  всё равно  работает без изменений  2 к 1.  Что я делаю не так?

GyverLibs, 2022-01-29:
Перед подключением библиотеки?

schpuntik, 2022-01-29:
ДА  Так  и есть.

schpuntik, 2022-01-29:
Всё разобрался.  Снова  невнимательность новичка.
Я прописал  перед  декларацией энкодера  но не  библиотеки.
Теперь прописал   перед библиотекой и всё  заработало .
Спасибо на наводку.

closed 2022-01-29 by schpuntik