GyverLibs/EncButton #22: Работа EncButton на матричной клавиатуре

Maxim0x00k, 2022-02-09:
Добрый день! Я заранее извиняюсь, если вопрос глупый, но я перед этим гуглил и искал и правда не нашел как внедрить двойной клик в матричную клавиатуру. В виртуальном режиме указано 

> Виртуальный режим позволяет получить все возможности библиотеки EncButton в ситуациях, когда кнопка не подключена напрямую к микроконтроллеру, либо для её опроса используется другая библиотека:
Аналоговая клавиатура (например через библиотеку [AnalogKey]
Матричная клавиатура (например через библиотеку [SimpleKeypad])"

Вопрос, как с помощью этой библиотеки сделать одну или несколько кнопок на матричной клавиатуре c двойным нажатием ?

GyverLibs, 2022-02-09:
привет, передавать актуальное состояние кнопок в tick. Объект библиотеки EncButton должен быть настроен в виртуальный режим, как во встроенном примере virtual

GyverLibs, 2022-02-09:
Ну, вообще абсолютно не так) добавлю примеры через часок

Maxim0x00k, 2022-02-09:
@GyverLibs Я буду вам дико благодарен!

GyverLibs, 2022-02-09:
закинул. Нужно обновить библиотеку, я сделал чтобы в виртуальном режиме уровень кнопки сразу был высокий, для более логичной работы

Maxim0x00k, 2022-02-09:
@GyverLibs Громадное вам спасибо!!!

closed 2022-02-09 by Maxim0x00k