GyverLibs/EncButton #25: Correct syntax for callback mode in version 2

Mark-81, 2022-03-12:
What is the correct syntax to enable the callbacks in version2?

```
#define  EB_FAST  30    
#define  EB_BETTER_ENC  
#define  EB_HALFSTEP_ENC

#define  EB_DEB  50      
#define  EB_STEP  100    
#define  EB_CLICK  400   
#define  EB_HOLD  1000   

#include <EncButton2.h>
EncButton2<EB_CALLBACK> enc(11, 9, 10);

void myRight() {
  Serial.println("RIGHT_HANDLER");
}
void myLeft() {
  Serial.println("LEFT_HANDLER");
}

void setup() 
{
  Serial.begin(115200);
  while(!Serial);

  enc.attach(RIGHT_HANDLER, myRight);
  enc.attach(LEFT_HANDLER, myLeft);
}
 
void loop() 
{
  enc.tick();
}
```

This leads to a "crash" of the firmware, the USB port disappear and nothing works. 
Instead, using version 1 of the library:

```
#include <EncButton.h>
EncButton<EB_CALLBACK, 11, 9, 10> enc;
```

Works fine.

GyverLibs, 2022-03-12:
what board are you using? I have stable work EncButton2 + EB_CALLBACK on ATmega328 now

Mark-81, 2022-03-12:
As said I'm using an Arduino Leonardo compatible board, so it's an ATmega32U4

GyverLibs, 2022-03-12:
I confirm, there is a problem with 32U4, will try to fix it today

GyverLibs, 2022-03-12:
Done, v1.21 works correctly!

closed 2022-03-12 by GyverLibs