GyverLibs/EncButton #26: Не работает "Полушаговый" энкодер

Denis20221, 2022-04-05:
Стоит "тот" самый энкодер, где надо два щелчка прокрутить что бы произошел один тик.
#define EB_HALFSTEP_ENC(Как по инструкции перед подключением либы), вешает наглухо энкодер так что даже обычная кнопка не работает.

Запустил голый пример из библиотеки под названием optimisation, и снова вставив EB_HALFSTEP_ENC полез смотреть в serial. Там при прокрутке энкодер чудит. Но стоит раз нажать кнопку, как отрабатывать начинает энкодер "правильно", но кнопка больше не реагирует...

GyverLibs, 2022-04-06:
Алгоритм `EB_HALFSTEP_ENC` не очень хорошо работает из за особенности таких энкодеров, у меня часть энкодеров работает  идеально, часть - иногда с ошибкой направления шага. Рекомендую использовать совместно с `EB_BETTER_ENC` - чуть тяжелее но отрабатывает чётко любые энкодеры.  
Но в реализации действительно был небольшой косяк, спасибо! Я релизнул v1.22, теперь энкодер работает более правильно и указанный пример реагирует на кнопку

closed 2022-04-06 by GyverLibs

Denis20221, 2022-04-06:
Благодарю! Обновил библиотеку и задействовал EB_HALFSTEP_ENC+EB_BETTER_ENC.
Все заработало.