GyverLibs/EncButton #29: Использование RX/TX (GPIO3/GPIO3) в качестве А и В в pinmode, A, B, KEY

brightproject, 2022-04-20:
Подскажите, использую ESP8266 в схеме управления котлом.
Все пины на вход и выход заняты.
_Остались GPIO9 и GPIO10, а также RX/TX (GPIO3/GPIO3)._
Эта библиотека не может опрашивать данные ноги на вход?
Пробовал и через:

> pinMode(1, FUNCTION_3);
> pinMode(1, INPUT);

Часть кода такая:

> #include <SPI.h>
> #include <EncButton2.h>
> EncButton2<EB_ENCBTN> enc(INPUT, 1, D2, D7);  // энкодер с кнопкой (A, B, KEY)
> void setup() {
>   pinMode(1, INPUT);
>   Serial.begin(115200);
> }

Ничего не происходит...
Не хотелось бы использовать мультиплексоры - все таки надо, чтобы устройство не потребляло много электричества и было отчасти "автономным".

GyverLibs, 2022-04-20:
Библиотека работает с любыми GPIO пинами, опрашивая их как digitalRead(). Дублировать пинмоды не нужно, они настраиваются в библиотеке.

По поводу системных пинов esp8266 смотрите документацию и схемотехнику конкретной платы, там много особенностей и ограничений

brightproject, 2022-04-20:
Благодарю за ответ, возможно, что у меня при использовании например пина RX - не выполняется его опрос из-за того, что он(GPIO9) не "подтянут" к земле или 3.3 вольтам?

GyverLibs, 2022-04-20:
Кнопку в принципе невозможно опросить, если пин болтается в воздухе, он обязательно должен быть подтянут противоположно подключению кнопки. Именно поэтому input pullup во всех примерах сделан

brightproject, 2022-04-20:
> Кнопку в принципе невозможно опросить, если пин болтается в воздухе

А энкодер?

GyverLibs, 2022-04-20:
А энкодер это физически просто две кнопки. Если используется голый энкодер - нужно подтягивать пины. Если используется китайский модуль - там уже всё подтянуто к vcc

brightproject, 2022-04-20:
> Если используется китайский модуль - там уже всё подтянуто к vcc
Использую "китайский" круглый, где подтянуто.
Но все равно не работает RX на вход...

GyverLibs, 2022-04-20:
Также нужно помнить, что прижатый к GND пин TX не даст esp запуститься, с энкодером - модулем такое возможно если позиция энкодера "не защелкнута"

brightproject, 2022-04-20:
> прижатый к GND пин TX не даст esp запуститься

ESP запускается, но когда в TX or RX находится какой либо контакт от энкодера, то прошивка не начинается.

GyverLibs, 2022-04-20:
Эм, сериал не будет работать, если на его ноги повесить энкодер...

GyverLibs, 2022-04-20:
Если хочется пинов сэкономить - нужно делать энкодер отключаемым на момент прошивки, а сериал - не запускать

brightproject, 2022-04-20:
> сериал - не запускать

Хочется еще и отладку делать)

GyverLibs, 2022-04-20:
Ну, значит есп32 брать

brightproject, 2022-04-20:
> Ну, значит есп32 брать

У нее при использовании всех пинов - это не мешает отладке?

GyverLibs, 2022-04-20:
У неё просто пинов раза эдак в 4 больше, хватит на любой каприз и ещё останется

brightproject, 2022-04-20:
Спасибо за подсказки.
Доброй ночи!

GyverLibs, 2022-04-20:
Удачи)

closed 2022-04-20 by GyverLibs

brightproject, 2022-04-23:
Подскажите, можно использовать сдвиговый регистр или мультиплексор для подключения энкодера?

GyverLibs, 2022-04-23:
можно, но это будет работать плохо или вообще не будет