GyverLibs/EncButton #31: Увеличение время срабатывания между тиками

brightproject, 2022-07-10:
Библиотеки - зло.
Когда используешь отдельно и в чистом коде все работает как часики, но стоит проекту обрасти другими библиотеками или даже просто кодом и начинается чехарда.
В общем имею код, который опрашивает два датчика AHT10(с родной библиотекой AHTxx) и DS18B20(чистый код без библиотеки), также данные о температурах выводятся на дисплей нокия 5110 с помощью библиотеки adafruit.
Помимо этого использую микросхему PCF8575 для увеличения портов, работающую по шине I2C как и AHT10.
В общем использую библиотеку EncButton2.h и когда код "в сборе" то тики влево/вправо не отрабатывают при нормальном вращении, если вращать энкодер туда/сюда медленнее чем обычно, как бы от одного щелчка к другому, или например вращать быстро вправо несколько тиков, а потом один влево то библиотека определит поворот вправо, и наоборот.
Пробовал играться настройками(но чувствую они не для этого случая)
// ========= НАСТРОЙКИ (можно передефайнить из скетча) ==========
#define _EB_FAST 30         // таймаут быстрого поворота
#define _EB_DEB 50          // дебаунс кнопки
#define _EB_CLICK 400	    // таймаут накликивания
Увеличивал/уменьшал таймайт быстрого поворота - результата ноль.
Когда использую чистый код для энкодера без библы гайвера, то срабатывание всегда.
Подскажите какие параметры "подкрутить" и вообще как понять, в чем проблема.
P.S. Железка, которую проектирую, собрала на печатной плате.
В ней я, дабы выиграть количество ножек, спецом использовал RT и TX ноги ESP12 под дисплей нокия 5110 и теперь не имею возможности использовать сериал монитор для отладки.
Вывожу конечно команды отработки на дисплей, что в принципе одно и тоже.
Но код с библиотекой гайвера не отрабатывает, хотя в том же проекте теплицы он шурует нормально.
У себя пробовал отключать опросы датчиков и разумеется сразу энкодер работал нормально, но когда не используешь библиотеку, на чистом коде все работает четко, но библиотека гайвера для большого проекта удобнее.
Мне как раз нужно вставлять обработчик в условия и менять параметры от этого.
Используя чистый код без библиотеки получается кода становиться много и его отслеживать сложно.

GyverLibs, 2022-07-10:
В загруженной программе энкодер нужно сажать на прерывания. По поводу настроек - они не для этого, все написано в документации

GyverLibs, 2022-07-10:
Библиотеки - не зло, зло - куча голого кода в программе. Работу с одним компонентом нужно оборачивать хотя бы в структуру и выносить в отдельный файл, иначе будет не программа а помойка

brightproject, 2022-07-10:
> Библиотеки - не зло, зло - куча голого кода в программе. Работу с одним компонентом нужно оборачивать хотя бы в структуру и выносить в отдельный файл, иначе будет не программа а помойка

По возможности весь код в структурах)функциях.
На днях не смог партию DS18B20 запустить нормально от родной далласовской библиотеки!
При температуре выше 52 градусов библиотека обрабатывала цифровой сигнал не корректно, и выдавала -127.
Пришлось чистый код писать и тогда заработало.
Да и раньше, когда веб-программированием был занят, тоже вечно одна библиотека на js конфликтует с другой, и некоторые проблемы нерешаемы, пока не напишешь свой "чистый" код обработчика.
P.S. Датчики конечно не оригинал, фейковые.
https://pikabu.ru/story/drop_temperaturyi_datchikov_ds18b20_9260431

GyverLibs, 2022-07-10:
Ну, это уже маразм и поможет только шапочка из фольги)) если один и тот же код внутри класса и вне класса работает по разному - это уже клиника...

GyverLibs, 2022-07-10:
по далласу
> мысли о некорректной работе родной библиотеки с китайскими датчиками

Всё с точностью до наоборот - это китайские датчики не работают с родной библиотекой, потому что родная библиотека написана по таймингам, которые указаны в даташите. У китайских 99% эти тайминги начинают плавать при повышении температуры. Значит в библиотеке можно подстроить тайминги под китайские подвальные чипы. 
> Пришлось чистый код писать и тогда заработало.

А в библиотеке код какой? Грязный что ли?) Это ведь просто смешно..

brightproject, 2022-07-10:
> по далласу
> 
> > мысли о некорректной работе родной библиотеки с китайскими датчиками
> 
> Всё с точностью до наоборот - это китайские датчики не работают с родной библиотекой, потому что родная библиотека написана по таймингам, которые указаны в даташите. У китайских 99% эти тайминги начинают плавать при повышении температуры. Значит в библиотеке можно подстроить тайминги под китайские подвальные чипы, вместо того чтобы писать "родная библиотека - зло". Детский сад
> 
> > Пришлось чистый код писать и тогда заработало.
> 
> А в библиотеке код какой? Грязный что ли?) Это ведь просто смешно..
Мы все учились понемногу, / Чему-нибудь и как-нибудь!

brightproject, 2022-07-10:
> У китайских 99% эти тайминги начинают плавать при повышении температуры. Значит в библиотеке можно подстроить тайминги под китайские подвальные чипы, вместо того чтобы писать "родная библиотека - зло".

Как вы знаете, у библиотеки далласа два файла - _DallasTemperature.h и DallasTemperature.cpp_
В первом файле _DallasTemperature.h_ есть определение кодов ошибок:
// Error Codes
#define DEVICE_DISCONNECTED_C -127
Во втором файле _DallasTemperature.cpp_
Вот что пишут в комментариях, в каких случаях мы получим -127:
```
// returns temperature in degrees C or DEVICE_DISCONNECTED_C if the
// device's scratch pad cannot be read successfully.
// the numeric value of DEVICE_DISCONNECTED_C is defined in
// DallasTemperature.h. It is a large negative number outside the
// operating range of the device
```
1. DEVICE_DISCONNECTED_C -127 будет если неправильно задан адрес датчика - этот случай исключаем, т.к. датчик один и адрес задан конкретно
2. DEVICE_DISCONNECTED_C - 127 будет если сработает аларм по высокой +125 или по низкой -55 температурам - это тоже маловероятно, т.к. датчик показывает комнатную Т=25 и после нагрева доходит до 52 и выдает дисконнект -127.
3. DEVICE_DISCONNECTED_C -127 будет если происходит неверная конвертация из "сырой" цифры в цельсий:
`// convert from raw to Celsius
float DallasTemperature::rawToCelsius(int16_t raw) {

	if (raw <= DEVICE_DISCONNECTED_RAW)
		return DEVICE_DISCONNECTED_C;
	// C = RAW/128
	return (float) raw * 0.0078125f;

}`
Мне кажется здесь что то зарыто, но какие здесь тайминги и вообще, можно ли их в этой библиотеке корректировать - вопрос!

GyverLibs, 2022-07-10:
В этой - нет, тайминги живут в OneWireб на базе которой работает DallasTemperature

brightproject, 2022-07-11:
В общем, не знаю что делать, энкодер будто заторможенный, когда в коде идет обработка данных с сенсоров.
https://pastebin.com/3mDjy95Y
Как добавить прерывания, или может быть в коде в секции loop setup как то не так последовательность выполнения кода выстроена?
Подскажите пожалуйста советом, чувствую программист с меня пока не ахти.
Проект встал из за "вялой" работы энкодера, в остальном все ОК - и данные с датчиков, и работа с расширителем портов PCF8575 и связь с mqtt.
Меню можно конечно пользоваться, но оооочень надо приноровиться к энкодеру - вращать как бы "на кончиках" пальцев - чувствуя момент перехода от одного тика к другому)))
Если вращать чуть медленнее или быстрее то не будет срабатывания.
Вообще не понимаю, почему так стал работать...

GyverLibs, 2022-07-11:
Программа написана некорректно. Несколько советов:
- Выводить данные на экран нужно либо по таймеру, либо по факту изменения данных, которые на него выводятся
- Даллас нужно опрашивать не чаще 1 раза в секунду (точнее 750мс на максимальном разрешении)
- В загруженной программе энкодер сажается на прерывания, примеры есть в библиотеке

brightproject, 2022-07-11:
> * В загруженной программе энкодер сажается на прерывания, примеры есть в библиотеке

tickISR - это пример прерываний?

GyverLibs, 2022-07-13:
> Код отформатировал, но он не особо изменился

Зато его легче воспринимать тому, кто привык работать с форматированным кодом

> Как это выводить данные на экран по таймеру, millis() - который?

Да. В текущем случае - и по таймеру (для термометра), и по факту изменения (по щелчку энкодера)

> Пины подключения энкодера A и B ножен к ардуине - 0 и 1 или D2 и D3? Или в принципе любые?

Так в вашем же сообщении выше это всё написано + есть ссылка на библиотеку....

> В загруженном или быдлокоде именно такое проявление

По разному может быть

closed 2023-08-20 by GyverLibs