GyverLibs/EncButton #32: Работоспособность кода на ESP8266/ESP12F

brightproject, 2022-10-08:
```
/*
   Пример работы с энкодером с прерыванием. Максимальная чёткость работы
   в любом быдлокоде!
*/

#define CLK 2
#define DT 3
#define SW 4

#include "GyverEncoder.h"
Encoder enc1(CLK, DT, SW);

void setup() {
  Serial.begin(9600);
  attachInterrupt(0, isrCLK, CHANGE);    // прерывание на 2 пине! CLK у энка
  attachInterrupt(1, isrDT, CHANGE);    // прерывание на 3 пине! DT у энка
}

void isrCLK() {
  enc1.tick();  // отработка в прерывании
}
void isrDT() {
  enc1.tick();  // отработка в прерывании
}

void loop() {
  enc1.tick();
  if (enc1.isRight()) Serial.println("Right");         // если был поворот
  if (enc1.isLeft()) Serial.println("Left");

  if (enc1.isRightH()) Serial.println("Right holded"); // если было удержание + поворот
  if (enc1.isLeftH()) Serial.println("Left holded");
}
```
Данный код перезагружает модуль ЕСП12 и выдает циклически ошибку

```
19:57:06.723 -> >>>stack>>>
19:57:06.723 -> 
19:57:06.723 -> ctx: cont
19:57:06.723 -> sp: 3fffff00 end: 3fffffc0 offset: 0000
19:57:06.769 -> 3fffff00:  feefeffe feefeffe feefeffe 3ffef314  
19:57:06.823 -> 3fffff10:  000000fe 00000000 00000000 00000000  
19:57:06.870 -> 3fffff20:  00000000 00000000 00000000 00ff0000  
19:57:06.923 -> 3fffff30:  5ffffe00 5ffffe00 00002580 3ffee588  
19:57:06.970 -> 3fffff40:  00000000 00000003 00000001 4020219e  
19:57:07.024 -> 3fffff50:  40100439 00000001 3ffee520 402021b0  
19:57:07.070 -> 3fffff60:  4020272c 00000003 00000001 402026b1  
19:57:07.124 -> 3fffff70:  401000f8 00000000 feefeffe feefeffe  
19:57:07.171 -> 3fffff80:  3fffdad0 00000000 3ffee574 40202750  
19:57:07.224 -> 3fffff90:  3fffdad0 00000000 3ffee574 40201072  
19:57:07.324 -> 3fffffa0:  feefeffe feefeffe feefeffe 40201d2c  
19:57:07.324 -> 3fffffb0:  feefeffe feefeffe 3ffe85d8 40100cf5  
19:57:07.425 -> <<<stack<<<
19:57:07.425 -> 
19:57:07.425 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
19:57:07.472 -> H!⸮⸮)⸮@H⸮⸮tH⸮ISR not in IRAM!
19:57:07.572 -> 
19:57:07.572 -> User exception (panic/abort/assert)
19:57:07.618 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
19:57:07.664 -> 
19:57:07.664 -> Abort called

```

Автор библиотеки пишет, что нужно в имя функции подставлять **IRAM_ATTR void имя(){}**
https://alexgyver.ru/lessons/interrupts/
Но в таком случае функция isrCLK не декларирована вообще.

brightproject, 2022-10-08:
Работающее решение на прерываниях для ESP8266
```
#define ENC_A 12
#define ENC_B 14
#define ENC_KEY 13

#include "GyverEncoder.h"
Encoder enc1(ENC_A, ENC_B, ENC_KEY);

void setup() {
  Serial.begin(9600);
  enc1.setType(TYPE2);        // тип энкодера TYPE1 одношаговый, TYPE2 двухшаговый. Если ваш энкодер работает странно, смените тип\=
  attachInterrupt(0, isr_ENC_A, CHANGE);    // прерывание на 2 пине! CLK у энка
  attachInterrupt(1, isr_ENC_B, CHANGE);    // прерывание на 3 пине! DT у энка
}

//void ICACHE_RAM_ATTR my_int();

ICACHE_RAM_ATTR  void isr_ENC_A() {
  enc1.tick();  // отработка в прерывании
}

ICACHE_RAM_ATTR void isr_ENC_B() {
  enc1.tick();  // отработка в прерывании
}

void loop() {
  enc1.tick();
  if (enc1.isRight()) Serial.println("Right");         // если был поворот
  if (enc1.isLeft()) Serial.println("Left");

  if (enc1.isRightH()) Serial.println("Right holded"); // если было удержание + поворот
  if (enc1.isLeftH()) Serial.println("Left holded");
}
```

GyverLibs, 2022-10-08:
читайте доки на свою версию SDK esp8266, там в разных версиях прерывания нужно декларировать по разному

closed 2022-10-08 by GyverLibs