GyverLibs/EncButton #33: ArduinoOTA не работает при вызове метода enc.tick()

borodihin, 2022-11-07:
Проект на esp32. Функция loop () выглядит примерно следующим образом:
```
void loop () {
  ArduinoOTA.handle();
  enc.tick();
  getSensors();
}
```
В случае вызова метода enc.tick() при работе стандартных процедур ArduinoOTA код в loop() продолжает выполняться.
Энкодер подключен без использования прерываний.

GyverLibs, 2022-11-07:
> код в loop() продолжает выполняться

а не должен?

borodihin, 2022-11-07:
Я так понимаю, что пока грузится обновление по воздуху - не должен. Во всяком случае, если tick() не вызываю - не выполняется.

GyverLibs, 2022-11-07:
К сожалению ничем не могу помочь, у меня ArduinoOTA отказывается работать на esp32. Что можно попробовать:
- Обновить ядро esp32
- Обновить ArduinoOTA
- Заменить `enc.tick();` на три штуки `digitalRead()` - по сути кроме чтения пина в тике ничего не происходит
- Включить отладку в ядре esp32, посмотреть что пишет
- Включить встроенную в ArduinoOTA обработку статусов как в примере ArduinoOTA (onStart, onEnd...)

GyverLibs, 2022-11-07:
Мне удалось завести ArduinoOTA, всё работает без проблем. Возможно стоит обновить ядро esp32, оно до сих пор в активной разработке и довольно криво работает.
[image]

closed 2022-11-07 by GyverLibs

borodihin, 2022-11-07:
getSensors() у меня выводит на дисплей значения сенсоров, в ArduinoOTA обработка статусов включена и на тот же дисплей выводится прогресс обновления. В другом месте программы по прерыванию таймера на дисплей выводится текущее время, поэтому прерывания по таймеру приходится выключать при начале обновления.
Если вызываю enc.tick() значения сенсоров отображаются на дисплее "поверх" статуса обновления. Если закомментировать - всё нормально.
В любом случае, спасибо за помощь!

GyverLibs, 2022-11-07:
Значит надо ковырять дальше, начать с этого
- Обновить ядро esp32
- Обновить ArduinoOTA
- Заменить enc.tick(); на три штуки digitalRead() - по сути кроме чтения пина в тике ничего не происходит
- Включить отладку в ядре esp32, посмотреть что пишет
- Посмотреть, какой статус ошибки появляется у ArduinoOTA

borodihin, 2022-11-07:
Чудеса:
если enc.tick();  вызываю в конце цикла loop() всё работает.
`void loop () {
  ArduinoOTA.handle();
  getSensors();
  enc.tick();
}`