GyverLibs/EncButton #34: Не помещается в ATtiny13

MrOzean, 2022-11-21:
Версия библиотеки: 2.0
Какой используется МК: ATtiny13A + [MicroCore](https://github.com/MCUdude/MicroCoreMicroCore)

Библиотека настроенная для полноценного энкодера с кнопкой [фото](https://github.com/GyverLibs/EncButton/raw/main/doc/encAli.png) не помещается во флеш память  ATtiny13 (https://github.com/GyverLibs/EncButton/raw/main/doc/encAli.png)

```cpp
#include <EncButton.h>
#include <stdbool.h>

#define ENCODER_KEY_PIN 2
#define ENCODER_ROT_RIGHT_PIN 3
#define ENCODER_ROT_LEFT_PIN 4
#define PWM_PIN 0

EncButton<EB_TICK, ENCODER_ROT_RIGHT_PIN, ENCODER_ROT_LEFT_PIN, ENCODER_KEY_PIN> enc;  

bool onOff = TRUE;

void setup() {
}

void loop() {
  if (enc.tick()) {
    if (enc.click())
      onOff = !onOff;
  }
}
```
Этот код занимает 1142 байт(111%) флеша.
Это баг библиотеки, выбранного ядра, или она в принципе не поддерживает камни меньше ATtiny85?

GyverLibs, 2022-11-21:
Видимо для всех возможностей библиотеки килобайта флэша будет маловато 😂 будет время - сделаю микро версию с более дешёвыми алгоритмами

MrOzean, 2022-11-22:
Эх, пошёл искать тогда библиотеку поменьше. Ты убери из описания ATtiny13, чтобы людей не смущать 😅

GyverLibs, 2022-11-22:
ну так то для тини13 заявлена скорость чтения пина, а не память) обработка кнопки скорее всего влезет

MrOzean, 2022-11-22:
Одной кнопкой сыт не будешь 😅

GyverLibs, 2022-11-22:
в принципе сами алгоритмы обработки занимают 600 байт, есть потенциал для тини-либы

GyverLibs, 2023-08-20:
новая версия вмещается =)

closed 2023-08-20 by GyverLibs