GyverLibs/EncButton #37: Проблема с энкодерами

kraa965, 2023-05-22:
Добрый день!
Подскажите, в чем может быть проблема, энкодеры реагируют хорошо, а вот кнопки не считываются вообще
`
#include <Arduino.h>
#include <EncButton2.h>
#include <U8g2lib.h>

const int NUM_SLIDERS = 4;
int analogSliderValues[NUM_SLIDERS] = {0};

EncButton2<EB_BTN> button_pin[NUM_SLIDERS];
EncButton2<EB_ENC> encoder[NUM_SLIDERS];

int knobState[NUM_SLIDERS];
int Mute[NUM_SLIDERS];
int knobVal[NUM_SLIDERS];
int mappedvalueslider[4];
int mappedvalueslider1[4];
long unsigned int remappedvalue[4];
int mappedvalue;

U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ A5, /* data=*/ A4);

void setup() {
    Serial.begin(115200);
    u8g2.begin();

    //--Кнопки (SW)--//
    button_pin[0].setPins(INPUT_PULLUP, 13);
    button_pin[1].setPins(INPUT_PULLUP, 3);
    button_pin[2].setPins(INPUT_PULLUP, 2);
    button_pin[3].setPins(INPUT_PULLUP, 12);

    //--Энкодер (CLK & DT)--//
    encoder[0].setPins(INPUT, 5, 4);
    encoder[1].setPins(INPUT, 7, 6);
    encoder[2].setPins(INPUT, 9, 8);
    encoder[3].setPins(INPUT, 11, 10);

    for (int i = 0; i < NUM_SLIDERS; i++) button_pin[i].tick();
    for (int i = 0; i < NUM_SLIDERS; i++){
      //button_pin[i].click();
      analogSliderValues[i] = 1023;
      knobState[i] = 0;
      Mute[i] = 0;
      encoder[i].counter = 51;
    }
}

void sendSliderValues() {
    String builtString = String("");
    for (int i = 0; i < NUM_SLIDERS; i++) {
        builtString += String((int)analogSliderValues[i]);
        if (i < NUM_SLIDERS - 1) {
            builtString += String("|");
        }
    }
    Serial.println(builtString);
}

void checkButtons() {
    for (int i = 0; i < NUM_SLIDERS; i++) {
      if (digitalRead(button_pin[i].click()) == 1) {
        knobState[i] = 0;
       }
       if (digitalRead(button_pin[i].click()) == 0 && knobState[i] == 0) {
        Mute[i] = !Mute[i];
        knobState[i] = 1;
        //delay(25);
      }
    }
}

void checkEncoders() {
  for (int i = 0; i < NUM_SLIDERS; i++) {
    knobVal[i] = encoder[i].counter;
    if (digitalRead(encoder[i].left()) || digitalRead(encoder[i].right()) && knobVal[i] > 0 && knobVal[i] < 102 && Mute[i] == 0) {
      analogSliderValues[i] = map(knobVal[i], 0, 102, 0, 1023);
      } else if (Mute[i] == 0 && (knobVal[i] >= 102)) {
        analogSliderValues[i] = 1023;
        encoder[i].counter = 102;
        } else if (Mute[i] == 1) {
          analogSliderValues[i] = 0;
          } else {
            analogSliderValues[i] = 0;
            encoder[i].counter = 0;
    }
  }
}

void display_draw() {
  int a = 0;
  //рисует цифры и (знаки %) рядом с 5 рамками
  for(int i = -10; i <= 62; i= i + 15) {
    u8g2.setFontPosTop();
    u8g2.setFont(u8g_font_6x10);
    u8g2.setCursor(0,i); //положение курсора
    u8g2.print(a);
    u8g2.setCursor(120,i);
    u8g2.print("%");
    a++;
  }

  //рисует 4 рамоки
  for(int b = 0; b <= 64; b= b + 17) {
    u8g2.drawFrame(8, b, 92, 12);
  }

  //заполняет каждый из 4 индикаторов выполнения их значениями
  for(int a = 0; a < 4; a++) {
      u8g2.drawBox(10, 2 + (17 * a), mappedvalueslider[a], 8);
  }

  //Выведите процентное значение каждого ползунка
  for(int i = 0; i < 4; i++) {
    if(remappedvalue[i] < 10)
      {
        u8g2.setCursor(113, 2 + (17 * i));
      }
      else if(remappedvalue[i] > 9 && remappedvalue[i] < 100)
      {
        u8g2.setCursor(107, 2 + (17 * i));
      }
      else if(remappedvalue[i] == 100)
      {
        u8g2.setCursor(101, 2 + (17 * i));
      }
      u8g2.print(remappedvalue[i]);
    }
}

void mappingvalues(){
  for(int a = 0; a < 4; a++){
      mappedvalueslider[a] = map(analogSliderValues[a], 0, 1023, 0 ,88);
      remappedvalue[a] = map(analogSliderValues[a], 0, 1023, 0, 100);
  }
}

void loop() {
  mappingvalues();
  u8g2.firstPage();
  do {
    for (int i = 0; i < NUM_SLIDERS; i++) encoder[i].tick();
    for (int i = 0; i < NUM_SLIDERS; i++) button_pin[i].tick();
    for (int i = 0; i < NUM_SLIDERS; i++) {
      knobVal[i] = encoder[i].counter;
    }
    sendSliderValues();
    checkEncoders();
    checkButtons();
    display_draw();
  } while (u8g2.nextPage());
}
`

kraa965, 2023-05-23:
Так же попытался перевести на библиотеку GyverOLED, но видимо что-то с моей функцией дисплея не так, вместо вывода 4 полос, заполняется весь экран

kraa965, 2023-05-29:
Совершенно не понимаю косяка

brightproject, 2023-06-11:
> Совершенно не понимаю косяка

Так, а вы вообще примеры то смотрели?
https://github.com/GyverLibs/EncButton#%D0%BF%D0%BE%D0%BB%D0%BD%D0%BE%D0%B5-%D0%B4%D0%B5%D0%BC%D0%BE-eb_tick
У вас в бесконечном цикле ересь какая то!

closed 2023-06-12 by kraa965