GyverLibs/EncButton #39: Добавить поддержку энкодера

Maxx1200, 2023-06-25:
можно ли добавить поддержку такого энкодера с такой логикой
[image]

Пример кода с использованием другой библиотеки

//просто пример
#include <Encoder.h> // https://github.com/P­aulStoffregen/Encode­r/releases/tag/1.4.3
 
//пины энкодера
#define PIN_A D5
#define PIN_B D6
 
//объект энкодера
Encoder encoder(PIN_A, PIN_B);
 
//предыдущее значение энкодера
long oldPosition = -999;
 
void setup() {
  //инициализация COM порта
  Serial.begin(115200);
}
 
void loop() {
  //считываем значе­ние от энкодера
  long newPosition = encoder.read();
 
  //если новое знач­ение не равно предыд­ущему,
  //то выводим его на COM порт
  if (newPosition != oldPo­sition) {
    oldPosition = newPosition;
    Serial.print("Encoder value: ");
    Serial.println(newPosition);
  } 
  delay(100); //задержка для стаб­илизации
}

GyverLibs, 2023-06-25:
это стандартный энкодер, библиотека его поддерживает

Maxx1200, 2023-06-25:
Пробовал вот так запустить через эту библиотеку но срабатывает через раз

// пример с библиотекой EncButton2

// Опциональные дефайн-настройки (показаны по умолчанию)
#define EB_FAST 30     // таймаут быстрого поворота, мс
#define EB_DEB 50      // дебаунс кнопки, мс
#define EB_CLICK 400   // таймаут накликивания, мс

#include <EncButton2.h>
EncButton2<EB_ENCBTN> enc(INPUT_PULLUP, D5, D6, D7);  // энкодер с кнопкой
// для изменения направления энкодера поменяй A и B при инициализации

void setup() {
  Serial.begin(115200);
    // настройки
  //enc.setButtonLevel(HIGH);     // уровень кнопки: LOW - кнопка подключает GND (по умолч.), HIGH - кнопка подключает VCC
  enc.setEncType(1);  // тип энкодера: EB_FULLSTEP (0) по умолч., EB_HALFSTEP (1) если энкодер делает один поворот за два щелчка
}

void loop() {
  enc.tick();                       // опрос происходит здесь

  // =============== ЭНКОДЕР ===============
  // обычный поворот
  if (enc.turn()) {
    // можно опросить ещё:
    Serial.println(enc.counter);  // вывести счётчик
    //Serial.println(enc.fast());   // проверить быстрый поворот
    Serial.println(enc.dir()); // направление поворота
  }

  if (enc.left()) Serial.println("L");     // поворот налево
  if (enc.right()) Serial.println("R");   // поворот направо
  if (enc.leftH()) Serial.println("LH");   // нажатый поворот налево
  if (enc.rightH()) Serial.println("RH"); // нажатый поворот направо

  // =============== КНОПКА ===============
  if (enc.press()) Serial.println("press");
  if (enc.click()) Serial.println("click");
  if (enc.release()) Serial.println("release");

  if (enc.held()) Serial.println("held");     // однократно вернёт true при удержании
  //if (enc.hold()) Serial.println("hold");   // будет постоянно возвращать true после удержания
  if (enc.step()) Serial.println("step");     // импульсное удержание

  // вывести количество кликов
  if (enc.hasClicks()) {
    Serial.print("has clicks ");
    Serial.println(enc.clicks);
  }
}

GyverLibs, 2023-06-25:
```cpp
enc.setEncType(EB_HALFSTEP);  // тип энкодера: EB_FULLSTEP (0) по умолч., EB_HALFSTEP (1) если энкодер делает один поворот за два щелчка
```

Maxx1200, 2023-06-25:
пробую так срабатывает каждые 4 поворота 
enc.setEncType(EB_FULLSTEP);

пробую так срабатывает через раз 
enc.setEncType(EB_HALFSTEP);

GyverLibs, 2023-06-25:
так, а где купить такой энкодер?

Maxx1200, 2023-06-25:
[оптический энкодер grayhill 62A.D.V.pdf](https://github.com/GyverLibs/EncButton/files/11859947/grayhill.62A.D.V.pdf)

GyverLibs, 2023-06-25:
беда. Пока ничем не могу помочь тогда

Maxx1200, 2023-06-25:
тут можно купить) хаахха)
https://www.chipdip.ru/product0/8013607824

GyverLibs, 2023-06-25:
хахха)

Maxx1200, 2023-06-25:
вот логика энкодера визуально!
pjrc.com/teensy/td_libs_Encoder.html
посмотреть, можно ли добавить такую логику!

GyverLibs, 2023-06-25:
что угодно можно добавить, но мне не на чем тестировать

Maxx1200, 2023-06-25:
Я бы мог бы тестировать)

Maxx1200, 2023-06-25:
Я тебе на почту написал посмотри пожалуйста

GyverLibs, 2023-07-09:
так это же обыкновенный энкодер, такой же как на круглой плате. Должен работать без изменения настроек в библиотеке. Попробуй вот этот код
```cpp
void setup() {
  Serial.begin(115200);
}

int counter;
void loop() {
  if (tick()) Serial.println(counter);
}

bool tick() {
  static uint8_t prev;
  static int8_t ecount;
  uint8_t state = digitalRead(2) | (digitalRead(3) << 1);
  if (prev != state) {
    ecount += ((0x49941661 >> ((state | (prev << 2)) << 1)) & 0b11) - 1;
    prev = state;
    
    Serial.println(String("count: ") + ecount);
    if (state == 0b11 && ecount) {
      counter += (ecount < 0) ? 1 : -1;
      ecount = 0;
      return 1;
    }
  }
  return 0;
}
```

Интересует что выведет при одном щелчке. Но это тот же алгоритм что в encbutton

GyverLibs, 2023-07-09:
так подожди. Кажется у тебя энкодер "щёлкает" в каждой позиции, а обычные китайские щёлкают каждые 4 позиции. Добавлю в новой версии, эт ерунда
[image]

GyverLibs, 2023-07-09:
сейчас можешь вот тут https://github.com/GyverLibs/EncButton/blob/e135656fc3e215f17bb0199e142b3eafb65773d8/src/EncButton2.h#L274C17-L274C17
оставить в скобках только `_ecount`

Maxx1200, 2023-07-09:
> так это же обыкновенный энкодер, такой же как на круглой плате. Должен работать без изменения настроек в библиотеке. Попробуй вот этот код
> 
> ```c++
> void setup() {
>   Serial.begin(115200);
> }
> 
> int counter;
> void loop() {
>   if (tick()) Serial.println(counter);
> }
> 
> bool tick() {
>   static uint8_t prev;
>   static int8_t ecount;
>   uint8_t state = digitalRead(2) | (digitalRead(3) << 1);
>   if (prev != state) {
>     ecount += ((0x49941661 >> ((state | (prev << 2)) << 1)) & 0b11) - 1;
>     prev = state;
>     
>     Serial.println(String("count: ") + ecount);
>     if (state == 0b11 && ecount) {
>       counter += (ecount < 0) ? 1 : -1;
>       ecount = 0;
>       return 1;
>     }
>   }
>   return 0;
> }
> ```
> 
> Интересует что выведет при одном щелчке. Но это тот же алгоритм что в encbutton

проверил все работает реагирует на каждое положение энкодера

Maxx1200, 2023-07-09:
> сейчас можешь [вот тут](https://github.com/GyverLibs/EncButton/blob/e135656fc3e215f17bb0199e142b3eafb65773d8/src/EncButton2.h#L274C17-L274C17) оставить в скобках только `_ecount`

Изменил, попробовал. тоже. реагирует на каждое положение энкодера.

GyverLibs, 2023-07-09:
Я Encbutton с нуля переписал, будет очень сильно лучше. Энкодер после ресета тоже брыкаться не будет

GyverLibs, 2023-08-20:
сё, можно пользоваться

closed 2023-08-20 by GyverLibs