GyverLibs/EncButton #40: Поддержка Arduino Nano 33 BLE

hz61p1, 2023-08-19:
При использовании Arduino Nano 33 BLE не распознаются повороты энкодера, только взаимодействие с кнопкой.
Энкодер EC11 на круглой плате.
Пробовал разные режимы INPUT и INPUT_PULLUP, не помогло.

hz61p1, 2023-08-19:
```
attachInterrupt(digitalPinToInterrupt(...), isr, CHANGE); 
attachInterrupt(digitalPinToInterrupt(...), isr, CHANGE); 
```
Не дает результата

hz61p1, 2023-08-19:
Полных скетч

```cpp
#include <EncButton.h>

#define EB_PIN_S1 D2
#define EB_PIN_S2 D3
#define EB_PIN_KEY D4

EncButton<EB_TICK, EB_PIN_S1, EB_PIN_S2, EB_PIN_KEY> enc;

void isr()
{
    enc.tickISR();
}

void setup()
{
    Serial.begin(115200);

    attachInterrupt(digitalPinToInterrupt(EB_PIN_S1), isr, CHANGE);
    attachInterrupt(digitalPinToInterrupt(EB_PIN_S2), isr, CHANGE);
}

void loop()
{
    if (enc.tick())
    {
        if (enc.turn())
            Serial.println("turn");

        if (enc.turnH())
            Serial.println("hold + turn");

        if (enc.left())
            Serial.println("left");

        if (enc.right())
            Serial.println("right");

        if (enc.leftH())
            Serial.println("leftH");

        if (enc.rightH())
            Serial.println("rightH");

        if (enc.press())
            Serial.println("press");

        if (enc.click())
            Serial.println("click");

        if (enc.held())
            Serial.println("held");

        if (enc.step())
            Serial.println("step");

        enc.resetState();
    }
}
```

RockKenny, 2023-08-20:
Тоже самое на esp32.
Использовал стандартный пример "демо"
Энкодер без внешних подтяжек и внешнего антидребезга.

GyverLibs, 2023-08-20:
Всё изумительно работает на esp8266 и esp32, в том числе на прерываниях, ещё раз сел проверил

hz61p1, 2023-08-20:
@RockKenny Я согласен с @GyverLibs, на esp32 все работает. Посмотрите закрытые ису, там было касаемо esp32.

hz61p1, 2023-08-20:
@GyverLibs Я также тестировал на разных платах у меня в наличие, это esp32,  esp8266, pro mini, в разных комбинациях и с 5V и 3.3V так же, на всех перечисленных работает, а на сабже нет. Возможно это из-за того что сабж работает на mbed os поверх которого arduino фреймворк. Так же проверял прерывания и они работают. Так как у меня нет jlink то остаются извращения по отладке)

RockKenny, 2023-08-20:
> @GyverLibs Я чет хз. Если ставлю инпутпулап, вообще не двигается. Если убираю, то энкодер произвольно сам постоянно срабатывает влево, хотя его никто не трогает.

GyverLibs, 2023-08-20:
Ну случайностей не бывает. Пины точно обычные gpio без подводных камней как у esp? Есть схема энкодера? Куда тянет при замыкании, есть ли внешняя подтяжка, или это китайский модуль стандартный

RockKenny, 2023-08-20:
[image]
Голый энкодер, без обвязки и подтяжек. Подключал по вашей схеме с сайта.
Пины такие (17, 4, 16); на счет подводных камней хз, но с энкбаттон2 работает идеально.

GyverLibs, 2023-08-20:
Тут нужен input pullup естественно. Алгоритм обработки такой же как в старой версии, в этом не должно быть проблемы. Какая платформа, плата, версия sdk?

RockKenny, 2023-08-20:
При INPUT_PULLUP повороты не работают, только кнопка.

Ардуино
Плата wroom 32 devkit v1
последняя версия https://github.com/espressif/arduino-esp32

GyverLibs, 2023-08-20:
Ну быть такого не может, я на этой же плате проверял - работает

RockKenny, 2023-08-20:
Я не троллю вас, точно!
Заливаю пример из второй версии библиотеки, все четко работает. Потом удалю эту библиотеку и закидываю 3 версию. Заливаю пример и дальше как я описал. Если есть INPUT_PULLUP, работает только кнопка, если нет, то само срабатывает.

GyverLibs, 2023-08-20:
Сейчас поищу голый энк соберу такую же схему попробую

GyverLibs, 2023-08-20:
Ой не знаю дядь, работает как часики, или даже лучше 
```cpp
EncButton eb(4, 17, 16, INPUT_PULLUP);
````

[image]

RockKenny, 2023-08-20:
Ну ладно, посижу на второй версии... Спасибо за труд.

GyverLibs, 2023-08-20:
Так не пойдет, надо разобраться в чем дело. Загрузи

```cpp
void setup() {
  Serial.begin(115200);
  pinMode(4, INPUT_PULLUP);
  pinMode(17, INPUT_PULLUP);
}
void loop() {
  Serial.println(digitalRead(4) | (digitalRead(17) << 1), BIN);
  delay(40);
}
```

Покрути энк покажи какие изменения в порте летят

GyverLibs, 2023-08-20:
Должно быть так
```
11
10
0
1
11
```
От щелчка до щелчка если убрать повторы

RockKenny, 2023-08-20:
Сейчас сделаю

GyverLibs, 2023-08-20:
А лучше так
```cpp
void setup() {
  Serial.begin(115200);
  pinMode(4, INPUT_PULLUP);
  pinMode(17, INPUT_PULLUP);
}
byte p;
void loop() {
  byte v = digitalRead(4) | (digitalRead(17) << 1);
  if (p != v) {
    p = v;
    Serial.println(v, BIN);
  }
}
```

RockKenny, 2023-08-20:
На первом примере у меня бесконечно летела 11, даже повернут не успевал. Со вторым примером работает.
Сделал 3 поворота влево, потом 3 вправо

1
0
10
11
1
0
10
11
1
0
10
11
10
0
1
11
10
0
1
11
10
0
1
11
10
11

GyverLibs, 2023-08-20:
Всё работает, открывай пример демо, заменяй объявление энкодера на
```cpp
EncButton eb(4, 17, 16, INPUT_PULLUP);
``` 
И вперёд

RockKenny, 2023-08-20:
Теперь понятно в чем проблема. Я по аналогии со второй версией писал так
`EncButton eb(INPUT_PULLUP, 4, 17, 16);`
В мануале к новой либо нет сноски об этом, либо яя проморгал.

вот так EncButton eb(4, 17, 16, INPUT_PULLUP); работает

GyverLibs, 2023-08-20:
Каком мануале, готовый пример же был, как можно было испортить пример 😭😭

closed 2023-08-20 by GyverLibs

RockKenny, 2023-08-20:
Спокойно, в родном примере нет ничего про правильный синтаксис INPUT_PULLUP, что его теперь в конце надо писать. Или не в этом была ошибка?

GyverLibs, 2023-08-20:
Указал режим работы вместо пина, видно ведь что это неправильно выглядит даже. В документации написано
```cpp
// настроить пины (энк, энк, кнопка, pinmode энк, pinmode кнопка)
EncButton(uint8_t encA, uint8_t encB, uint8_t btn, uint8_t modeEnc = INPUT, uint8_t modeBtn = INPUT_PULLUP);
```

И никто никого не обвиняет! У нас тут максимально дружеская атмосфера, всё на приколе и для души

RockKenny, 2023-08-20:
Смайлы со слезами, я подумал что у вас истерика. Ну вы сказали, что я пример испортил, а я только вбил свои пины и по аналогии со второй версией пулап вначале поставил. Но косяк мой, просмотрел ибо искал специальную сноску типа "вот что поменялось не забудьте исправить". Но вы не обязаны так делать, сказано, что не совместимо, значит надо читать внимательно. Спасибо, что помогли.

GyverLibs, 2023-08-20:
Да, краткий гайд по миграции на v3 я забыл добавить, сделаю

hz61p1, 2023-08-21:
@GyverLibs ису закрыта по невнимательности? Ису то была не про esp32))

GyverLibs, 2023-08-21:
Ису была про старую версию библиотеки, если ТС заявит о себе - откроем

hz61p1, 2023-08-21:
@GyverLibs Обновил либу, и поворот теперь работает, но только если не использовать прерывания, а без них могут быть пропуски если основный цикл чем-то занят

GyverLibs, 2023-08-21:
Значит неправильно используются или настроены прерывания на конкретной платформе. Библиотека не делает в прерывании ничего интереснее чем обычный digitalRead

GyverLibs, 2023-08-21:
@hz61p1 тогда давай проверять как выше мы есп32 проверяли
```cpp
void setup() {
  Serial.begin(115200);
  pinMode(4, INPUT_PULLUP);
  pinMode(17, INPUT_PULLUP);
}
byte p;
void loop() {
  byte v = digitalRead(4) | (digitalRead(17) << 1);
  if (p != v) {
    p = v;
    Serial.println(v, BIN);
  }
}
```

Нужно покрутить энк и написать сюда что летит в порт

GyverLibs, 2023-08-21:
Ну давай проверим теперь энкодер в прерывании
```cpp
#include <Arduino.h>

void isr() {
    Serial.println(digitalRead(2) | (digitalRead(3) << 1), BIN);
}

void setup() {
    Serial.begin(115200);
    attachInterrupt(0, isr, CHANGE);
    attachInterrupt(1, isr, CHANGE);
}

void loop() {
}
```

От щелчка до щелчка должно выводить 
```
11
1
0
10
11
```

hz61p1, 2023-08-24:
@GyverLibs сериал не работает в прерывании, но я уже забил на энкодер и перешел на датчик жестов

GyverLibs, 2023-08-24:
Очень странно, как они так криво написали сериал что он не работает...