GyverLibs/EncButton #41: Опрос кнопок в момент перезагрузки устройства

brightproject, 2023-08-23:
Возможно сделать опрос состоянии кнопок в цикле `setup(){}`?
Что то вроде такого
```
void setup() {
	btn.setButtonLevel(HIGH);
    Serial.begin(115200);
    btn.tick();
    if (btn.click()) {
        Serial.println("Правая кнопка удержана при перезагрузке");
    }
}
```
У меня этот код не заработал в setup.
Код [индусов](https://github.com/m5stack/M5Stack/blob/master/src/utility/Button.h) свободно опрашивает кнопку, если ее удерживать во время перезапуска МК, путем нажатия RESET на отладочной плате.
```
  if (Btn.read())
    Serial.begin(115200);
    Serial.println("Правая кнопка удержана при перезагрузке");
    }
```

GyverLibs, 2023-08-23:
Всё верно! Чтение кнопки производится с учётом настроенного уровня кнопки:
```cpp
    bool read() {
        return EBread(pin) ^ read_bf(EB_INV);
    }
```

brightproject, 2023-08-24:
Все хорошо, но мне очень не хватает debounce в секции `setup(){}`
Инициализирую кнопку так
`Button btn_left(39, INPUT);// внешняя подтяжка
`
И применяю
```
setup(){
      if (btn_left.read()) 
	  {
        rsel++;
        if (rsel >= ROWS) rsel = 0;
      }
}
```
Переменная rsel почти всегда увеличивается на 2(два) значения, видимо кнопка определяет нажатие и при НАЖАТИИ и при ОТПУСКАНИИ.
Можно, что то придумать, или именно с этой библиотекой применение без `tick()` в бесконечном цикле - неполноценно?