GyverLibs/EncButton #42: Проект на wokwi с кнопками EncButton

brightproject, 2023-08-27:
Использую библиотеку кнопок, для некоторых проектов, которые есть в "железе".
Над проектом работают несколько человек, и у некоторых нет доступа к настоящим микроконтроллерам, дисплеями и кнопкам.
Как писать код спрашивается?
На помощь приходит онлайн-симулятор wokwi.
Но к сожалению, библиотека [EncButton](https://github.com/GyverLibs/EncButton) там не является "по умолчанию".
Решение - написать класс отдельным файлом, как это было предложено [здесь.](https://wokwi.com/projects/373591584298469377)
Я попробовал реализовать такой же файл, но у меня две кнопки, и я их инициализирую так
```
  pinMode(34, INPUT_PULLUP);
  pinMode(35, INPUT_PULLUP); 
```
Ввиду того, что ограничения wokwi видимо срабатывают, когда слишком много файлов добавляешь - не дает сохранить.
Но удалось скачать архив, из 16-ти файлов.
Можно их вручную создать и тогда проект скомпилируется.
[image]
Вы сможете вручную установить проект на wokwi и подсказать, что не так с кнопками?
[Flightinstruments_TFT_eSPI.zip](https://github.com/GyverLibs/EncButton/files/12448649/Flightinstruments_TFT_eSPI.zip)
Буду вам безмерно благодарен!
p.s. bool read(); не реализована в файле btn.h и поэтому не смог [сделать](https://github.com/GyverLibs/EncButton/issues/41#issuecomment-1690110385) опрос кнопок в секции `setup(){}`

GyverLibs, 2023-08-27:
библиотека некорректно работает в wokwi по непонятным причинам

brightproject, 2023-08-27:
> библиотека некорректно работает в wokwi по непонятным причинам

Например строка 998 *ino файла
```
      if (btn_left.hold(1))//вход в меню параметров
      {
        myMenus.menuState = MENU_LIST;
        myMenus.loadPrefs = true;

        preferences.begin ("huVVer-app", false); // restore brightness during setup
        Brightness = preferences.getUInt("Brightness", 1023);
        myFlight.Backlight(Brightness);
        preferences.end();

        if (DEBUG) Serial.println ("Menu List");
      }
```
На реальном устройстве нажимаю один раз кнопку, подключенную к пину 34 и потом удерживаю, и библиотека кнопки понимает действие и код переходит к меню параметров.
И так далее с другими комбинациями.
На wokwi же, что то не так.

GyverLibs, 2023-08-27:
В Wokwi содержатся все библиотеки из официального реестра Arduino. EncButton в ней тоже и она абсолютно такая же, как на гитхабе
[image]

```cpp
#include <EncButton.h>
Button b(3);

void setup() {
  Serial.begin(115200);
}

void loop() {
  b.tick();
  if (b.hold()) Serial.println(0);
  if (b.hold(1)) Serial.println(1);
}
```

brightproject, 2023-08-27:
> В Wokwi содержатся все библиотеки из официального реестра Arduino. EncButton в там тоже есть и она абсолютно такая же, как на гитхабе

Странно, добавлял EncButton в список библиотек wokwi, но при компиляции ругалось, что нет класса **Button**
`Button btn_left(39, INPUT);`
Хотя на компьютере для настоящего устройства все компилируется без проблем.

GyverLibs, 2023-08-27:
у wokwi какие то проблемы с подключением EncButton, почему то иногда ругается. Не знаю почему

GyverLibs, 2023-08-27:
вот сейчас раз 20 скомпилил с изменениями, без ошибок всё

GyverLibs, 2023-08-27:
> P.S. bool read(); не реализована в файле btn.h и поэтому не смог https://github.com/GyverLibs/EncButton/issues/41#issuecomment-1690110385 опрос кнопок в секции setup(){}

так и не должна. Это копия класса VirtButton, он к пину не привязан

GyverLibs, 2023-08-27:
так, ну он частично откуда то тянет старую версию библиотеки. Сейчас перезалью релиз видимо

[image]

brightproject, 2023-08-27:
> так и не должна. Это копия класса VirtButton, он к пину не привяза

Сначала я немного затупил, увидев класс `btn.h,` хотел обычной библиотекой работать с кнопками...
Вот и начал импровизировать...

GyverLibs, 2023-08-27:
я перезалил, ждём когда реестр обновится

GyverLibs, 2023-08-27:
[вот тут](https://www.arduinolibraries.info/libraries/enc-button) можно мониторить

brightproject, 2023-08-27:
> я перезалил, ждём когда реестр обновится

Да, понял, ждем.
Интересно, что на реальном устройстве библиотеку использую так
```
#include <EncButton.h>
Button btn_left(39, INPUT);// внешняя подтяжка
Button btn_right(36, INPUT);// внешняя подтяжка
void setup()
{
  // таймаут удержания (кнопка)
  btn_left.setHoldTimeout(1000); 
  btn_right.setHoldTimeout(1000); 
  // таймаут гашения дребезга кнопки (кнопка)
  btn_left.setDebTimeout(20);
  btn_right.setDebTimeout(20);
}
void loop()
{
	
    btn_left.tick();
    btn_right.tick();
}
```
А на `wokwi` пока не объявить пины
```
#include <EncButton.h>
Button btn_left(39, INPUT);// внешняя подтяжка
Button btn_right(36, INPUT);// внешняя подтяжка
void setup()
{
  // объявление пинов на вход с внешней подтяжкой
  pinMode(34, INPUT);
  pinMode(35, INPUT);
  // таймаут удержания (кнопка)
  btn_left.setHoldTimeout(1000); 
  btn_right.setHoldTimeout(1000); 
  // таймаут гашения дребезга кнопки (кнопка)
  btn_left.setDebTimeout(20);
  btn_right.setDebTimeout(20);
}
void loop()
{
	
    btn_left.tick();
    btn_right.tick();
}
```
Компиляция с ошибкой.

brightproject, 2023-08-27:
> [вот тут](https://www.arduinolibraries.info/libraries/enc-button) можно мониторить, ждём версию 3.0.1

Заметно так "подросла" библиотека с 15 кБ до 440 кБ 😀

GyverLibs, 2023-08-27:
это я гифки нечайно завернул. 3.0.1 весит 44 кб

brightproject, 2023-08-27:
Когда объявляем кнопку
`Button btn_left(34, INPUT);// внешняя подтяжка`
То класс, по умолчанию устанавливает подтяжку внутреннюю `INPUT_PULLUP`?
Если не указать явно.

```
class Button : public VirtButton {
   public:
    Button(uint8_t npin = 0, uint8_t mode = INPUT_PULLUP) {
        init(npin, mode);
        setButtonLevel(LOW);
    }

    // указать пин и его режим работы
    void init(uint8_t npin = 0, uint8_t mode = INPUT_PULLUP) {
        pin = npin;
        pinMode(pin, mode);
    }

    // прочитать текущее значение кнопки (без дебаунса)
    bool read() {
        return EBread(pin) ^ read_bf(EB_INV);
    }

    // функция обработки, вызывать в loop
    bool tick() {
        return VirtButton::tick(EBread(pin));
    }

   private:
    uint8_t pin;
};
```
Где то у wokwi несостыковка реальной электроники и модели в части подтяжки кнопок.
Поэтому и подтупливает симулятор.

GyverLibs, 2023-08-27:
Нечего там не подтупливает, я же кидал выше скриншот. По умолчанию класс Button это input pullup и low. Если у кнопки внешняя подтяжка к gnd - то настраивать нужно input и high

brightproject, 2023-08-27:
> Нечего там не подтупливает, я же кидал выше скриншот. По умолчанию класс Button это input pullup и low. Если у кнопки внешняя подтяжка к gnd - то настраивать нужно input и high

Вот в симуляторе делаю кнопку без резистора, т.е. без "внешней" подтяжки в 3.3 вольт.
`Button btn_left(34);// по умолчанию внутренняя подтяжка`
Но если так указать, то симулятор выдает ложные срабатывания.
Когда ставлю резисторы к питанию 3.3 вольт, то кнопку уже надо инициализировать так
`Button btn_left(34, INPUT);// внешняя подтяжка`
Но компилятор на компьютере соберет прошивку, а wokwi, будет ругаться, пока в секции setup не укажу пины явно.
```
void setup()
{
  pinMode(34, INPUT_PULLUP);
}
```
Проверил это утверждение в симуляторе и на компиляторе ПК только что.

GyverLibs, 2023-08-27:
Я писал уже выше - у wokwi какие то проблемы с библиотекой, поэтому ждём перезалива

brightproject, 2023-08-27:
> Я писал уже выше - у wokwi какие то проблемы с библиотекой, поэтому ждём перезалива

Да, большое спасибо за помощь.
На настоящем железе библиотека работает хорошо.
Вы можете сделать пример для wokwi как [этот](https://wokwi.com/projects/373591584298469377)
Только, чтобы использовал не `btn.h,` а `EncButton`?
Сейчас в симуляторе использую пока файл `btn.h`, но там нет функции read() для прямого чтения кнопки.

GyverLibs, 2023-08-27:
Я же написал - ждём обновления библиотеки в реестре. Если нужна реализация кнопки независимо от библиотеки как в примере - нужно создать в симуляторе файл и закинуть в него содержимое файлов virtbutton и button, убрав лишние include

brightproject, 2023-08-27:
Что-то и правда wokwi не адекватно работает с ядром библиотеки.
Если указать подтяжку пина кнопки как "внутреннюю"
`pinMode(3, INPUT_PULLUP);`
То компилятор выдаст ошибку

[image]

Если указываем подтяжку пина кнопки как "внешнюю"
`pinMode(3, INPUT);`
То код скомпилируется, правда схема без резистора подтяжки к +3.3 вольт будет работать не корректно.
Делал попытки зарегистрироваться в Discord, чтобы в клубе worki написать о проблеме кнопок, и еще про работу с симулятором с ПК через VS Code, но что-то не смогу попасть в Discord...попробую еще днем.

GyverLibs, 2023-08-27:
тут проблема скорее в том, что вовка не всегда может найти библиотеку. Это чисто случайное явление, у меня сейчас один и тот же код то компилится то нет
[image]

brightproject, 2023-08-28:
> тут проблема скорее в том, что вовка не всегда может найти библиотеку. Это чисто случайное явление, у меня сейчас один и тот же код то компилится то нет [image]

С ошибкой компиляции, тут вероятно проблема шлюза wokwi с интернетом - не всегда он "подтягивает" библиотеки с гитхаба.
Сами разработчики пишут, что решение - перейти на платный аккаунт, и там можно свои, персональные библиотеки грузить...но что то сомнительная услуга.
Еще надо учесть такой момент - что симулятор, работает медленнее чем реальное железо, и в крупных проектах, где в бесконечном цикле лууп, помимо опроса кнопок еще и другой код находиться, нужно увеличить время опроса клика(подбирается опытным путем).
https://wokwi.com/projects/374296311592617985
Данное issue больше относиться к нюансам wokwi, нежели к библиотеке EncButton.

GyverLibs, 2023-08-28:
Возможно библиотека стала тяжелее из за картинок и Вовка не хочет ее качать, попробую картинки отдельно залить куда нибудь в след обновлении

brightproject, 2023-08-28:
> Возможно библиотека стала тяжелее из за картинок и Вовка не хочет ее качать, попробую картинки отдельно залить куда нибудь в след обновлении

А по части прерываний, примеры нашел только для энкодера, для кнопок не работает.
https://wokwi.com/projects/374296311592617985

GyverLibs, 2023-08-28:
В документации всё максимально подробно написано. Прерывания только для энкодера

brightproject, 2023-08-28:
> В документации всё максимально подробно написано. Прерывания только для энкодера

Кажется есть понимание.
Но, несмотря на обновление библиотеки, тот же platformIO пока видит версию 3.0.0, видимо у каждого ресурса свои периоды запуска CRON для обновления библиотек.
[image]

GyverLibs, 2023-08-28:
Тем не менее третья версия "иногда" работает, не думаю что дело в этом

brightproject, 2023-08-28:
> Тем не менее третья версия "иногда" работает, не думаю что дело в этом

Хорошие новости, в первую очередь для себя - удалось запустить wokwi на ПК в VS Code.
Скорость работы существенно выше, и с библиотеками пока нет проблем.
[image]

GyverLibs, 2023-08-28:
да, так будет работать, потому что PIO качает файл с гитхаба, компилирует, и уже скомпиленный бинарник симулируется в вовке

GyverLibs, 2023-08-29:
Юрич разрулил, всё воркает

closed 2023-08-29 by GyverLibs

brightproject, 2023-08-31:
Все равно, в симуляторе не могу заставить код, с опросом кнопок через библиотеку EncButton
https://wokwi.com/projects/374518066429236225
Когда нет задержек в коде программы(в идеальной программе в "вакууме"), тогда все работает, и установленные пресеты на время опроса при клике(или hold'е) отрабатывают как надо.
Как только устанавливаешь имитацию загруженной(настоящей) программы, с минимальной задержкой 500 мсек, сразу все перестает работать как следует.
Уже клик нужно удерживать не 50 мс, как в пресете
`btnLeft.setClickTimeout(50);`
А около полсекунды, чтобы tick() в бесконечном цикле, смог опросить кнопку.
Пример прерывания

> https://github.com/GyverLibs/EncButton#кнопка-1
Вовсе не работает.

GyverLibs, 2023-08-31:
Если в программе есть задержки в 500 мс, то эту программу нужно стереть и написать заново. Особенно с учётом того что у есп32 есть не только freertos, а два блин аппаратных ядра

brightproject, 2023-08-31:
> Если в программе есть задержки в 500 мс, то эту программу нужно стереть и написать заново. Особенно с учётом того что у есп32 есть не только freertos, а два блин аппаратных ядра

Да это симулированная программа, чтобы прерывания потестировать - но и она не завелась.

GyverLibs, 2023-08-31:
> setClickTimeout

Устанавливает тайм-аут, который библиотека ожидает следующего клика. Внимательнее к документации

brightproject, 2023-08-31:
> > setClickTimeout
> 
> Устанавливает тайм-аут, который библиотека ожидает следующего клика. Внимательнее к документации

До дыр доки затер.
В примере
https://wokwi.com/projects/374518066429236225
код чистый, задержка на миллис, составляет 50 мсек, таймаут клика 100 мсек.
Но опрос кнопок нестабильный, непредстказуемый.

GyverLibs, 2023-08-31:
Значит заклеить дырки и затирать заново. В доке по русски черным по белому написано - вызывать tick в основном цикле программы. В приложенном коде тик вызывается по таймеру. Это не будет работать, это неправильно, это бред)

brightproject, 2023-08-31:
> Значит заклеить дырки и затирать заново. В доке по русски черным по белому написано - вызывать tick в основном цикле программы. В приложенном коде тик вызывается по таймеру. Это не будет работать, это неправильно, это бред)

В цикле вообще не работает.
Вы посмотрите пример кода и все сами поймете.

GyverLibs, 2023-08-31:
Пример некорректный, я же написал

GyverLibs, 2023-08-31:
> ваша библиотека не работает в реальных проектах

Так не бывает. Если программа написана не одним местом - у неё нет причин не работать. У меня нет времени подробно изучать несколько тысяч строк кода, но при беглом осмотре уже обнаруживаются очень серьёзные ошибки. Например:

```cpp
(btn_left.hold() && btn_right.hold())
```
Шанс выполнения этого условия - один на  миллион, потому что `hold` - это событие, оно срабатывает **однократно**. Чтобы это условие сработало - кнопки нужно нажать в один и тот же такт процессора, что физически невозможно.

Далее, если я правильно вижу, то имеет место быть конструкция:
```cpp
void loop() {
  button.tick();
  if (button.....)

  // ......

  button.tick();
  if (button.....)
}
```
Это категорически неправильно. Ещё раз к документации: все события сбрасываются после вызова `tick`. Это означает, что события, полученные после первого вызова tick, после второго tick уже будут сброшены, и наоборот. Так нельзя, это ведь очевидно.

Не исключено что есть и другие ошибки. Так что это не наша библиотека не работает, это ваш код очень сильно кривой.  А почему работает другая библиотека для кнопки - потому что в ней по другому реализована логика

GyverLibs, 2023-08-31:
вообще страшно представить, как можно поддерживать проект с 2 тысячами строк в loop. Хоть бы на функции его разбили бы, не говорю уже о файлах. Это же просто жесть

brightproject, 2023-08-31:
> вообще страшно представить, как можно поддерживать проект с 2 тысячами строк в loop. Хоть бы на функции его разбили бы, не говорю уже о файлах. Это же просто жесть

Код основной программы работает с библиотекой от [m5stack](https://github.com/m5stack/M5Stack/blob/master/src/utility/Button.h)
Я усложнил задачу, использую 2 кнопки у устройства, когда у того же m5stack используется 3 и 4 кнопки, в зависимости от устройства.
Посему, мне для управления сложным меню, двух кнопок не хватает, при использовании с "простым" опросом кнопок `digitalRead();`
Поэтому и пришел к библиотеке `EncButton`

А вообще, программа написана далеко не программистом, но достаточно успешно работает на реальных авиаприборах и прошла много полетов.
Мне же нужно опрашивать кнопки, и т.к. их всего две у устройства, это накладывает свои приблемы.

brightproject, 2023-08-31:
> Далее, если я правильно вижу, то имеет место быть конструкция:
> 
> ```c++
> void loop() {
>   button.tick();
>   if (button.....)
> 
>   // ......
> 
>   button.tick();
>   if (button.....)
> }
> ```
> 
> Это категорически неправильно. Ещё раз к документации: все события сбрасываются после вызова `tick`. Это означает, что события, полученные после первого вызова tick, после второго tick уже будут сброшены, и наоборот. Так нельзя, это ведь очевидно.

Так и реализовано в коде
viod loop()   
{
    btn_left.tick();
    btn_right.tick();
    button1.loop();
    button2.loop();

      // Вход в меню настроек - удерживаем левую кнопку 3 секунды
      // if (btn_left.hold(1))//вход в меню параметров
      if(button1.isPressed())
      {
        myMenus.menuState = MENU_LIST;
        myMenus.loadPrefs = true;

        preferences.begin ("huVVer-app", false); // restore brightness during setup
        Brightness = preferences.getUInt("Brightness", 1023);
        myFlight.Backlight(Brightness);
        preferences.end();

        if (DEBUG) Serial.println ("Menu List");
      }
}
Только `EncButton` не может опросить клик и удержание, а `ezButton` с легкостью.
Да и нативная библиотека от m5stack опрашивала, и даже в секции setup(){} определяла кнопки...

GyverLibs, 2023-08-31:
> нативная библиотека от m5stack опрашивала, и даже в секции setup(){} определяла кнопки...

EncButton тоже в setup опрашивает кнопки, уже многократно обсуждали выше

brightproject, 2023-08-31:
> никаких проблем нет, если разобраться как работает библиотека и как нужно писать код с учётом этих особенностей

Так ведь разобрался, в простых примерах работает.
Добавляешься по правилам в нагруженный код и все ... приплыли.

GyverLibs, 2023-08-31:
Алло, я выше указал на очень грубые ошибки, это уже не "по всем правилам"! Encbutton работает абсолютно корректно в реальной программе любой сложности, если следовать логике и документации, это блин обработка кнопки всего лишь

GyverLibs, 2023-08-31:
> Только EncButton не может опросить клик и удержание

Пример есть? Потому что всё работает отлично

brightproject, 2023-08-31:
> > Только EncButton не может опросить клик и удержание
> 
> Пример есть? Потому что всё работает отлично

Так этот проект
https://wokwi.com/projects/374228486620365825
Строка `960` сейчас использует опрос синей кнопки библиотекой `ezButton` 
Нажатие определяется, с легкостью.
Раскомментируйте  строку `959` и только чудом можно угадать с какой продолжительностью кликнуть и удерживать синюю кнопку.

GyverLibs, 2023-08-31:
Так по этому проекту я написал выше - код написан неправильно, и указал где

brightproject, 2023-08-31:
Благодарю вас за безмерное терпение, мне уже самому надоел это разговор, но...вы не совсем правы, на счет неправильности кода.
Вернее не во всех моментах вы правы.
Во первых, код, размещенный в симуляторе
https://wokwi.com/projects/374228486620365825
Это почти точь-в-точь код, который находится в настоящем, физически-сделанном устройстве.
В котором тоже две кнопки, одна с прямой логикой 1-0, вторая с обратной 0-1.
И на настоящем "железе" код работает прекрасно, и библиотека `EncButton` работает, по крайней мере мне удалось добиться вменяемой работы и отработки тех нажатий, что я планировал.
Вопросы именно к симулятору wokwi и работы в нем обработчика кнопок.
Вам написал, с той позиции, что вы писали библиотеку, и знаете какие тайминги как она "крутиться" и работает.
Для меня библиотека по сути "черный ящик", я ее подключил, вызвал и хочу, чтобы она делала то, что может.
В недра ее конечно влазил, но без глубин познаний.
И путем сравнения, вижу, что три библиотеки работают нормально в симуляторе, а `EncButton` не хочет.
Ваше время готов компенсировать, поддержкой каналу и вашим трудам.
Библиотека и так, вероятно, будет находиться в недрах реального устройства, чтобы работать с кнопками.
Просто у меня несколько программистов удаленно и не имеют доступа к физическим устройствам.
И работу с проектом можно вести только через симулятор...но я никак не могу наладить обработку кнопок в симуляторе.
Надеюсь сейчас стало понятно, что за "сыр-бор"?

GyverLibs, 2023-08-31:
Нет, понятно не стало. То на что я указал выше - это критические  ошибки, из за них программа не работает. Я буду повторять это столько раз, сколько нужно, потому что это факт

brightproject, 2023-08-31:
В железке работает...с другими либами для кнопок работает ... код неверный.
Да в мире все ПО написано кодом, который в той или иной степени не верный.
Дело в "настройке" этих инструментов, чтобы оркестр заиграл😀

GyverLibs, 2023-08-31:
Вы надо мной сейчас издеваетесь?

GyverLibs, 2023-08-31:
И ещё один довод в пользу издевательства. Вы сравниваете обработку **нажатия** кнопки у других библиотек с **клик + удержание** у encbutton. Тут вроде не нужно быть программистом, чтобы понимать разницу между этими событиями. Тогда уж пишите encbutton - метод `press()`, а то сравнили тёплое с мягким. Но работать оно тоже будет некорректно из за критической ошибки в коде, о которой я писал выше

GyverLibs, 2023-08-31:
Также я ради интереса протестировал - loop в этом проекте выполняется ~450 миллисекунд. Скорее всего что то написано очень криво, и это называется не "реальная" программа - а очень плохая программа. Простым выходом из этой ситуации является разделение программы по разным ядрам есп32, например на одном ядре выполняется постоянная перерисовка дисплея, на втором - обработка всяких событий и gui. Иначе сделать отзывчивую работу с кнопками будет невозможно, 500мс это раз в 100 хуже чем должно быть

brightproject, 2023-09-01:
> Также я ради интереса протестировал - loop в этом проекте выполняется ~450 миллисекунд. Скорее всего что то написано очень криво, и это называется не "реальная" программа - а очень плохая программа. [...]

Ух, я тоже делал замеры быстродействия в бесконечном цикле, еще когда начинал переписывать код проекта.
Вам спасибо, что продолжаете мне отвечать😀
На самом деле, ~450 миллисекунд, измеренных вами, это виноват симулятор.
На реальном есп32-wroom все работает и крутиться +-согласно таймингам.
Ваша библиотека отрабатывает, и все нажатия проходят.
Я спокойно перемещаюсь по меню по всем case-switch условиям.
Наверное, пока с wokwi не разрешить вопрос быстродействия, обработчик EncButton не будет работать нормально.
Проект тяжелый, и это факс - вы посмотрите сколько графических спрайтов он прогоняет через loop.
При этом я получаю около 16 fps, а это очень добротный показатель для микроконтроллера в работе с графикой такого порядка.

GyverLibs, 2023-09-01:
> хотелось использовать весь функционал библиотеки EncButton.

для этого нужно исправить критическую ошибку, о которой я писал выше

brightproject, 2023-09-01:
> > хотелось использовать весь функционал библиотеки EncButton.
> 
> для этого нужно исправить критическую ошибку, о которой я писал выше

Да нет там критической ошибки, это типовой код, написанный для ардуино-совместимой платформы есп32.
Вы попробуйте проект контроллера теплицы в симуляторе запустить, библиотека EncButton тоже не справиться и там.
Не смотря, на правильно написанный код проекта и каноны распределения задач.

GyverLibs, 2023-09-01:
> Да нет там критической ошибки

Может мне большими буквами написать, тогда будет понятно? **В ПРОГРАММЕ ДОПУЩЕНА КРИТИЧЕСКАЯ ОШИБКА ПРИ РАБОТЕ С БИБЛИОТЕКОЙ ENCBUTTON, ИЗ ЗА КОТОРОЙ БИБЛИОТЕКА НЕСТАБИЛЬНО ОБРАБАТЫВАЕТ КНОПКУ.** Не нужно мне писать про какой-то "типовой" код, не существует такого понятия. Что это вообще за позиция такая? Разработчик библиотеки лично отвечает - в программе ошибка в таком то месте, вы ему пишете - нет там ошибки. Самому то не смешно? Я больше не собираюсь тратить своё время. Желаю успехов в этом непростом деле.

> библиотека EncButton тоже не справиться и там

библиотека EncButton замечательно работает в любом коде, который написан с учётом описанных в документации требований