GyverLibs/EncButton #43: Todo?

GyverLibs, 2023-08-27:
- setButtonLevel -> setBtnLevel?
- Буферизировать dir и fast?

brightproject, 2023-08-27:
Если можно, то в класс кнопки `class Button : public VirtButton {}`
добавить чтение в секции setup с дебонсом, что то наподобие
```
    bool read_debounce() {
        static uint32_t ms;
        static uint8_t pinVal;

        ms     = millis();
        pinVal = digitalRead(pin);
        if (_invert != 0) pinVal = !pinVal;
        if (ms - _lastChange < _dbTime) {
            _lastTime = _time;
            _time     = ms;
            _changed  = 0;
            return _state;
            // return EBread(pin) ^ read_bf(EB_INV);
        } else {
            _lastTime  = _time;
            _time      = ms;
            _lastState = _state;
            _state     = pinVal;
            if (_state != _lastState) {
                _lastChange = ms;
                _changed    = 1;
                if (_state) {
                    _pressTime = _time;
                }
            } else {
                _changed = 0;
            }
            return _state;
            // return EBread(pin) ^ read_bf(EB_INV);
        }
    }
```
Пробовал поэкспериментировать - но как то криво работало у меня это.
Оставил как было
```
    bool read() {
        return EBread(pin) ^ read_bf(EB_INV);
    }
```

GyverLibs, 2023-08-27:
Нет, это ерунда какая то

GyverLibs, 2023-08-27:
Если по каким то причинам нужно именно остаться в сетап, без примера с флагом как я писал ранее, то можно воспользоваться состоянием busy и погрузиться в цикл
```cpp
void setup() {
  do {
    btn.tick();
    if (btn.click()) foo();
  } while (btn.busy());
}
```

GyverLibs, 2023-08-27:
Есть нужно две и более кнопок затащить в обработку при старте - соответственно 
```cpp
while (b1.busy() || b2.busy() || b3.busy());
```

brightproject, 2023-08-28:
> Нет, это ерунда какая то. Опять же чтобы подобный код работал - его нужно вызывать циклично, в сетап это не будет работать

Вы как адепт ардуино, все же создали многие библиотеки "универсальными", работающими и с другими МК.
Сейчас популярный чип это esp32-s3 и teensy 4.1
Используя вашу библиотеку, еще версии 2, столкнулся с тем, что на esp32 она работала не совсем так как надо, особенно с энкодерами.
Стал разбираться, и оказалось, разработчики esp32 аппаратно сделали поддержку энкодера, назвали ее PCINT - то есть прерывания аппаратные.
Вы не думали реализовать поддержку PCINT в своей библиотеке?

GyverLibs, 2023-08-28:
> сделали поддержку энкодера, назвали ее PCINT

PCINT это обычные прерывания с пина, к энкодеру это никакого отношения не имеет... Библиотека изначально сделана универсальной для работы с прерываниями на любых платформах, в прерывании нужно вызвать специальный тикер. К слову PCINT есть и у МК AVR

GyverLibs, 2023-08-28:
Это библиотека энкодера от esp32, там же написано `rotary_encoder.h`. Encbutton не будет совмещаться с другими фреймворками и от них зависеть, это не имеет смысла. Encbutton замечательно интегрируется с любой платформой, потому что функция опроса помещается в обработчик прерывания

closed 2023-09-07 by GyverLibs