GyverLibs/EncButton #44: Migration from V2 to  V3:  tick() returns unreliable status

ShaggyDog18, 2023-09-11:
Unsuccessfully tried to migrate from V2 to v3 with troubles.  Platform: STM32F401
I use the following to handle events/actions:
```
if( inputEnc.tick() ){
        // event handler
        bool fastRotate = inputEnc.fast();
  	if( inputEnc.right() ) {
           // do something for encoder turning right
        }
        ....
  	if( inputEnc.left() ) {
           // do something for encoder turning left
        }
 }
```
So, it seems that tick() is skipping events (returns true for some events, does not return true for some events even though an event is happening); as a result, the handler does not process events, behaves as event did not happen... It skips events unregularly, so, I did not notice any pattern.  Should be noticed, I also call `tickISR()` in the interrupt for any pin state CHANGE. 
Please, check up if tick() is always returning `true` for any event.
Thank you.

GyverLibs, 2023-09-11:
Do you have only one `tick` function in loop?

ShaggyDog18, 2023-09-11:
yes, I do: just one `tick()` is in the loop() for encoder with button. I also have a separate button with a dedicated `button.tick()`  in the loop()...

GyverLibs, 2023-09-11:
no, v3 is more stable than v2, I'm sure that you do something wrong. Try examples from library. And give me a minimal example with this bug. Because tick always returns `true` on events, there can't be event without true result

closed 2023-09-13 by GyverLibs