GyverLibs/EncButton #46: Unusual step counting by rotary encoder

MisiekDP, 2023-10-13:
Hey,
I am using KY-040 rotary encoder on ESP8266 microcontroller in my project. I am using the EncButton library.
The problem I am experiencing is incorrect step counting when moving the encoder knob. When I turn the knob one time, the step is not counted, only the second move of the encoder counts one step. Each one step is counted as 2 shifts of the encoder knob. Does anyone know how I can change this so that each one shift of the knob counts one encoder step?

GyverLibs, 2023-10-13:
Hi, read the docs and configure library for your model of encoder

MisiekDP, 2023-10-13:
Ok, thanks for reply, I read this but I don't know how to set this type. For example I did this in my project and I get error like "there are no setEncType expression". How to set the model of encoder?
[image]

GyverLibs, 2023-10-13:
Maybe you have not up to date library version? Or some redefined macro with same name. Check demo example[demo example](https://github.com/GyverLibs/EncButton/blob/main/examples/demo/demo.ino)  of EncButton in your environment

GyverLibs, 2023-10-13:
Nice