GyverLibs/EncButton #47: Многократное срабатывание виртуальной кнопки при однократном нажатии кнопки

vopvivy, 2023-10-28:
Arduino Uno
Arduino IDE 2.1.1
EncButton v3.4

[image]
[MultiKey_Test.zip](https://github.com/GyverLibs/EncButton/files/13195494/MultiKey_Test.zip)

GyverLibs, 2023-10-28:
Внимательнее к [документации](https://github.com/GyverLibs/EncButton#%D0%BE%D0%B1%D1%80%D0%B0%D0%B1%D0%BE%D1%82%D0%BA%D0%B0-%D0%B8-%D0%BE%D0%BF%D1%80%D0%BE%D1%81)

> `[событие]` - функция вернёт `true` однократно при наступлении события. Сбросится **после следующего вызова функции обработки**

GyverLibs, 2023-10-28:
но согласен, во встроенном примере работы с матричной клавиатурой код написан некорректно, его нельзя б рать за основу. Пример переделаю

closed 2023-10-28 by GyverLibs

GyverLibs, 2023-10-28:
нужно настроить нулевой дебанус, чтобы кнопки срабатывали сразу

`btn.setDebTimeout(0);`

а также тикать нулевым значением в начале loop

`btn.tick(0);`

тогда при опросе клавиатуры кнопка получит актуальное значение и сразу просигналит о клике, а на следующей итерации loop сбросится обратно