GyverLibs/EncButton #49: Ke-040 и 74hc165

IljaGolenev, 2023-11-23:
Добавьте пожалуйста пример, как подключать энкодеры и кнопки через сдвиговый регистр 74hc165. Я вроде догадываюсь, что это через VirtEncButton, но как?

GyverLibs, 2023-11-23:
Через виртуал, и передавать в тикер актуальные значения с чипа расширителя

IljaGolenev, 2023-11-23:
> Через виртуал, и передавать в тикер актуальные значения с чипа расширителя

Угум... А как?
Здесь https://github.com/GyverLibs/EncButton/issues/4#issuecomment-888615885
Вы говорите что есть пример работы с pcf8574 в папке с примерами, только вот не нашел

GyverLibs, 2023-11-23:
Ну как, найти библиотеку для своего расширителя, и отправлять сигналы с кнопок в тик. Не понимаю какие тут могут быть вопросы)

IljaGolenev, 2023-11-23:
> Ну как, найти библиотеку для своего расширителя, и отправлять сигналы с кнопок в тик. Не понимаю какие тут могут быть вопросы)

Нашел на форуме пример
https://community.alexgyver.ru/threads/virtualnyj-ehnkoder-encbutton-i-sdvigovyj-vxodnoj-registr-74hc165.6953/

Переписал, но как-то странно работает. Это из-за wokwi или всё-таки меня тупого?)
https://wokwi.com/projects/381893607653627905

GyverLibs, 2023-11-23:
в wokwi стоит работать в такой последовательности:
- Проверить работает ли EncButton с их энкодером
- Проверить корректно ли работает расширитель пинов, кнопки на него навесить, энкодеры, покрутить понажимать
- И тогда уже соединять

IljaGolenev, 2023-11-26:
Библиотека прекрасно работает без регистра с их энкодером. С регистром, библиотекой, и кнопками все работает. Только вот когда пытаюсь подключить энкодер, то все ломается.
https://wokwi.com/projects/382402220201019393
(оставлю тут, вдруг поможет кому)

GyverLibs, 2023-11-26:
нууууу как бы это, это изначально не будет работать
[image]

IljaGolenev, 2023-11-27:
да и без нее криво...

GyverLibs, 2023-11-27:
Без нее хотя бы будет работать, если всё правильно закодить. С такой задержкой это исключено

closed 2024-01-25 by GyverLibs