GyverLibs/EncButton #52: Передать аргументы в событие.

jackfrombb, 2024-01-10:
Здравствуйте.
Во первых, спасибо за работу)

Теперь к делу. При реализации энкодера в классе, приходится делать метод события статическим, что лишает достпа к локальным переменным класса. Если будет  такая возможность, то добавьте передачу аргументов в событие.

Сам я не хотел бы пул реквестить в библиотеку, без нормального понимания внутренней структуры и достаточной компетенции в плюсах.

GyverLibs, 2024-01-10:
Привет, каких аргументов? Не очень понятна ситуация, нужно подробности

jackfrombb, 2024-01-10:
> Привет, каких аргументов? Не очень понятна ситуация, нужно подробности

Вот пример. 
Только сразу договоримся, что "метод" = "функция"  и наоборот, точно не знаю как в плюсах принято) 
Предположим, надо определить реализацию контроля от виртуального класса.
Присоеденить не статический метод класса мы не сможем, из за несовпадения типов.
[image]

Но, если мы сделаем метод статическим, то поля класса становятся недоступны, что логично
[image]

Я вижу такое, возможное, решение:

Делаем событие с аргументами, например, в стиле WindowsForms)
[image]

Чуть изменяем привязку события
[image]

И вызов
[image]

Тогда в классе можно сделать так:
[image]

GyverLibs, 2024-01-10:
а зачем вообще использовать attach в классе, если всё равно вызывается тикер и action можно опросить там же?..

GyverLibs, 2024-01-10:
также можно использовать лямбда-функцию

jackfrombb, 2024-01-10:
> а зачем вообще использовать attach в классе, если всё равно вызывается тикер и action можно опросить там же?..

В этом случае, при некоторых обстоятельствах, очень глючит. В частности, когда прерывания забирают много веремени у loop.

closed 2024-01-10 by jackfrombb

GyverLibs, 2024-01-10:
> очень глючит

не может такого быть)) Приаттаченный хэндлер вызывается из тикера
```cpp
    bool tick(bool s) {
        clear();
        s = pollBtn(s);
#ifndef EB_NO_CALLBACK
        if (cb && s) cb();
#endif
        return s;
    }
```

то есть `if (enc->tick()) myHandler();` это абсолютно тот же самый код получится

jackfrombb, 2024-01-10:
> > очень глючит
> 
> не может такого быть)) Приаттаченный хэндлер вызывается из тикера
> 
> ```c++
>     bool tick(bool s) {
>         clear();
>         s = pollBtn(s);
> #ifndef EB_NO_CALLBACK
>         if (cb && s) cb();
> #endif
>         return s;
>     }
> ```
> 
> то есть `if (enc->tick()) myHandler();` это абсолютно тот же самый код получится

Ок. Значит тогда глючило из за того что не вызывал тик в прерывании) Значит, вполне годное решение, спасибо ещё раз)
```
/// @brief Прерывание для обработки пропущенных считываний энкодера
/// @param args = NULL
/// @return nothing
bool IRAM_ATTR encTick(void *args)
{
    enc.tickISR();
    return false;
}
```