GyverLibs/EncButton #55: Внутрення подтяжка пинов INPUT_PULLUP

tiffass, 2024-02-16:
Не могу понять работает ли эта библиотека с внутренней подтяжкой пинов?
В примере demo написано
EncButton eb(2, 3, 4, INPUT); // + режим пинов энкодера
Меняю INPUT на INPUT_PULLUP  и ничего не работает. Точнее в одном из положений энкодера в мониторе порта видно что постоянно идет информация о вращении энкодера. Тип энкодера **eb.setEncType(EB_STEP2);**
Пробовал разные варианты 
EncButton eb(2, 3, 4, INPUT_PULLUP);
EncButton eb(2, 3, 4, INPUT_PULLUP,INPUT_PULLUP);
EncButton eb(2, 3, 4, INPUT_PULLUP,INPUT_PULLUP,INPUT_PULLUP);
и даже pinMode(2, INPUT_PULLUP); pinMode(3, INPUT_PULLUP);
результат одинаковый.
С внешней подтяжкой все нормально.

GyverLibs, 2024-02-16:
Всё работает отлично, внимательнее к схеме своего энкодера

closed 2024-02-16 by GyverLibs

tiffass, 2024-02-16:
Какая может быть схема? 3 проводка?

GyverLibs, 2024-02-16:
Мне то откуда знать? Голый это энкодер или модуль, если модуль то какой конкретно, и так далее

tiffass, 2024-02-16:
Модуль KY-040
[image]

но резисторы подтяжки с платы выпаяны, т.к. в дальнейшем планирую использовать голый энкодер. CLK подключен к пину 2, DT - 3. Кнопка не подключена т.к. не нужна. GND соответственно к  GND ардуины. 
В мониторе такая картина

[image]

С запаянными резисторами и подключенным "+" - все нормально

GyverLibs, 2024-02-16:
Можно прописать вручную пинмод input pullup на пины энкодера в setup(). Также возможно похожая проблема была в одной из прошлых версий gyverio, так что первым делом нужно обновить и енкбатн и gyverio до актуальных версий. Если не заработает - через несколько дней буду у компьютера, возьму такой же модуль, проверю. Какой микроконтроллер?

tiffass, 2024-02-16:
> Можно прописать вручную пинмод input pullup

Пробовал - не помогает

> обновить и енкбатн и gyverio до актуальных версий

С этого и начал

> Какой микроконтроллер?

Arduino uno - mega328

tiffass, 2024-02-16:
Хм ... полтергейст какой-то, попробовал еще раз прописать pinmodе input_pullup - заработало. А с EncButton eb(2, 3, 4, INPUT_PULLUP) так и не хочет

GyverLibs, 2024-02-16:
Запряг человека протестировать, он перегрел энкодер когда выпаивал резики) пока вопрос открытый

tiffass, 2024-02-16:
> а если добавить

Так работает, даже без bool EB_read(uint8_t pin).

> перегрел энкодер когда выпаивал

Газовой горелкой что-ли выпаивал? :)

GyverLibs, 2024-02-17:
Феном.. 

видимо дело в gyverio режиме пина, буду разбираться

GyverLibs, 2024-02-17:
всё пофиксил, спасибо за репорт

tiffass, 2024-02-17:
> пофиксил

Да, теперь работает