GyverLibs/EncButton #58: Добавить возможность игнорирования событий до спуска кнопки

lo1ol, 2024-07-10:
Добрый день,

Для демонстрации проблемы опишу пример. У меня есть кнопка и энкодер. При клике на кнопку происходит одно действие, при зажатии кнопки и вращении энкодера -- другое:

```cpp
void loop() {
    if (gBtn.pressing() && gEncoder.turn())
        doSomething1();

    if (gBtn.click())
        doSomething2();
}
```

Проблема в том, что если я быстро нажму на кнопку и в это время крутану энкодер, то отработают два действия. Когда хотелось бы, чтобы обработало одно. Я пробовал звать clear и  reset на кнопке в первом условии. Но это не помогает. 

Я ничего пока лучше не придумал, кроме как написать класс-обертку, имел такой метод. Есть ли какой-то стандартный способ для борьбы с таким поведением? Если нет, то хотелось бы иметь какой-то метод типа "ignoreTillPress", который игнорировал бы дальнейшие события, до тех пор пока зажата кнопка. т.к. уже несколько раз сталкивался с подобной проблемой

GyverLibs, 2024-07-10:
Привет, поворот энкодера сбрасывает некоторые события с кнопок, в ридми должно быть написано. Press/release не сбрасываются

lo1ol, 2024-07-11:
События же сбрасываются только на кнопке, которая привязана к `EncButton`? Не на всех? У меня кнопка и энкодер разделены просто.

Вот пример для воспроизведения:
```cpp
#define EB_NO_FOR
#define EB_NO_CALLBACK
#define EB_NO_COUNTER
#define EB_NO_BUFFER
#define EB_DEB_TIME 50    // таймаут гашения дребезга кнопки (кнопка)
#define EB_CLICK_TIME 200 // таймаут ожидания кликов (кнопка)
#define EB_STEP_TIME 200  // таймаут импульсного удержания (кнопка)
#define EB_HOLD_TIME 500  // таймаут удержания (кнопка)
#define EB_FAST_TIME 15   // таймаут быстрого поворота (энкодер)

#include <EncButton.h>

EncButton gEncoder(2, 3);
ButtonT<A0> gBtn;

void isr() {
    gEncoder.tickISR();
}

void setup() {
  Serial.begin(9600);
  attachInterrupt(0, isr, CHANGE);
  attachInterrupt(1, isr, CHANGE);
  gEncoder.setEncISR(true);
}

void loop() {
  gBtn.tick();
  gEncoder.tick();

  // При быстром повороте и нажатии выведется и kek1 и kek2
  if (gBtn.pressing() && gEncoder.turn())
    Serial.println("kek1");

  if (gBtn.click())
    Serial.println("kek2");
}
```

Вот такую обертку написал, чтобы решить проблему:
```cpp
template<uint8_t Pin>
class SmartButton : public ButtonT<Pin> {
public:
    void tick() {
        ButtonT<Pin>::tick();
        if (m_ignoreTillPressing && ButtonT<Pin>::press()) {
            m_ignoreTillPressing = false;
            m_timeout = millis();
            ButtonT<Pin>::clear();
        }

        // we need some some before wrapped button return click() == true, after press() == false
        if (!m_ignoreTillPressing && m_ignore && static_cast<uint16_t>(millis()) - m_timeout > EB_DEB_TIME) {
            m_ignore = false;
            ButtonT<Pin>::clear();
        }
    }

    void ignoreTillPressing() {
        m_ignoreTillPressing = true;
        m_ignore = true;
    }

    bool click() { return !m_ignore && ButtonT<Pin>::click(); }

private:
    bool m_ignoreTillPressing = false;
    bool m_ignore = false;
    uint16_t m_timeout;
};
```

GyverLibs, 2024-07-11:
А с чего бы энкодеру сбрасывать события с классов и объектов, о существовании которых он не знает?)) для этого и есть encbutton, который работает одновременно и с кнопкой и с энкодером

lo1ol, 2024-07-11:
> А с чего бы энкодеру сбрасывать события с классов и объектов, о существовании которых он не знает?)) 

Я это понимаю. Поэтому и предложил добавить какой-нибудь отдельный метод для этого. Просто у меня в проекте на самом деле не одна кнопка, а несколько. И кнопка, которая приходится зажимать -- это не та же самая кнопка, что и на энкодере физически расположена. Да и в целом я вижу проекты, где, например, у тебя есть куча кнопок, которые отвечают за разные настройки. И чтобы менять эти настройки -- ты должен нажимать на них и крутить энкодер, который расположен отдельно. При этом отдельный клик по этим кнопкам может означать что-то другое

Хорошо. Приведу еще один пример, где я столкнулся с аналогичной проблемой. Тут уже все проще и без энкодера и всего одна кнопка. У меня есть таймер, и запускать я хотел бы его по клику (возможно человек зажмет его, потом понял, что зажал случайно, подержит чуть-чуть и отпустит). Спешить при старте в моей задаче некуда.  А останавливать как только регистрируем нажатие (чем раньше мы зарегистрируем нажатие, тем точнее):

```cpp
        if (gStartBtn.press() && gTimer.state() == Timer::RUNNING) {
            gTimer.pause();
            gStartBtn.clear();
        }

        if (gStartBtn.click() && gTimer.state() ==Timer::STOPPED) {
                gTimer.start(m_printTime);
        }
```

тут так же отработают два действия, когда мне хотелось бы их разделить.

Btw. Еще такой вопрос. Почему, когда мы быстро нажимаем на кнопку, то событие click приходит не одновременно с событием release. click приходит с некоторым запозданием. Я бы возможно не гундел и просто спокойно написал обертку, которая делает то, что мне нужно. Но как оказалось, такая вот очевидная обертка не решает мою проблему:

```cpp
template<uint8_t Pin>
class SmartButton : public ButtonT<Pin> {
public:
    void tick() {
        m_ignoreTillPressing &= ButtonT<Pin>::press();
        ButtonT<Pin>::tick();
    }

    void ignoreTillPressing() { m_ignoreTillPressing = true; }
    bool click() { return !m_ignore && ButtonT<Pin>::click(); }

private:
    bool m_ignoreTillPressing = false;
};
```

Проблема здесь в том, что после того, как мы получим `press()` и наш флаг игнорирования сброится, click мы получим только через несколько тактов. И выходит, что мой флаг проигнорируется. Даже если tick так написать, то это не помогает:
```cpp
    void tick() {
        ButtonT<Pin>::tick();
        if (m_ignoreTillPressing && !ButtonT<Pin>::press()) {
            m_ignoreTillPressing = false;
            ButtonT<Pin>::clear()
        }
    }
```

GyverLibs, 2024-07-11:
Событие клик при

closed 2024-07-11 by GyverLibs

lo1ol, 2024-07-11:
> Событие клик приходит после обработки дебаунса, как и release. Если это не нужно - дебаунс можно установить в 0

хм. Мне кажется, что deb у них одновременно дорабатываться должен. Или у них разные таймеры на проверку этого? 

Просто в моем понимании приход события release однозначно говорит о том, что кнопку отпустили. И дополнительно к этому можно узнать отпустили ее после короткого клика, или после удержания (или не то и не то)

lo1ol, 2024-07-11:
ааа. Или release до обработки deb приходит?

GyverLibs, 2024-07-11:
События генерируются по очереди с каждым вызовом тик. Сначала общий release, затем release удержаний, потом клики

GyverLibs, 2024-07-11:
В принципе есть возможность сделать ignore. Свободных битов под новые флаги нет, но по сути именно эту роль выполняет флаг удержания энкодера, он является сигналом к игнорированию событий кнопки. Сделаю, скину протестировать

GyverLibs, 2024-07-11:
в общем вот так
```cpp
    void ignoreTilRelease() {
        bf.set(EB_EHLD);
    }
```
я пока не у компьютера

lo1ol, 2024-07-11:
Спасибо большое! Решение работает)

GyverLibs, 2024-07-11:
отлично, тогда залью обновку

lo1ol, 2024-07-11:
а обновите версию либы на platformio?) Или она сама через какое-то время автоматически выложится?