GyverLibs/EncButton #62: Энкодер type2

fergq, 2024-11-02:
Извиняюсь сразу за возможно тупой вопрос, но в голову уже мысли не приходят, а многое вообще не понимаю..

Как указать что у меня энкодер второго типа чтобы он стал работать правильно? (https://alexgyver.ru/encoder/?ysclid=m30krqqfh8518972050 - " setType(type); // тип энкодера TYPE1 одношаговый, TYPE2 двухшаговый. Если ваш энкодер работает странно, смените тип ")

Ещё раз извинюсь если вопрос туповат

GyverLibs, 2024-11-02:
Как в ридми/примерах

fergq, 2024-11-02:
всё перепробовал - серовно не корректно работает (

.left вообще никак не хочет выдавать

GyverLibs, 2024-11-02:
https://github.com/GyverLibs/EncButton?tab=readme-ov-file#%D1%82%D0%B8%D0%BF-%D1%8D%D0%BD%D0%BA%D0%BE%D0%B4%D0%B5%D1%80%D0%B0

fergq, 2024-11-02:
а, упс.. я пины попутал, спасибо большое

closed 2024-11-02 by fergq