GyverLibs/EncButton #64: Заменить два прерывания на одно в примерах

lo1ol, 2024-12-09:
Привет,

Делаю устройство, у которого есть энкодер и одна кнопка. Это устройство может уходить в сон (спасибо твоей либе GyverPower!) и я хотел реализовать подъем по вращению энкодера или по нажатию на кнопку.

По началу думал, что это не возможно т.к. в примерах для обработки пребываний с энкодера использованы два прерывания (а на Pro Mini их всего два можно привязать), но изучив что шлется по двум пинам энкодера, понял что и одного должно быть достаточно. Проверил -- так и есть)

Думаю стоит отметить этот момент в документации)

GyverLibs, 2024-12-09:
Привет, не понял ничего

lo1ol, 2024-12-09:
Сейчас пример для работы с пребываниями такой:

```cpp
EncButton eb(2, 3, 4);

void isr() {
  eb.tickISR();
}
void setup() {
  attachInterrupt(0, isr, CHANGE);
  attachInterrupt(1, isr, CHANGE);
  eb.setEncISR(true);
}
void loop() {
  eb.tick();
}
```

при этом для успешной обработки пребываний достаточно:

```cpp
EncButton eb(2, 3, 4);

void isr() {
  eb.tickISR();
}

void setup() {
  attachInterrupt(digitalPinToInterrupt(2), isr, CHANGE);
  eb.setEncISR(true);
}

void loop() {
  eb.tick();
}
```

и так можно сэкономить еще одно пребывание, повесив его, например, на кнопку для пробуждения

GyverLibs, 2024-12-09:
Нет, для полноценной обработки одного прерывания недостаточно. Также у аврок можно поднять прерывание на любом пине через pcint и не экономить

closed 2024-12-09 by GyverLibs