GyverLibs/EncButton #66: Оптический декодер.

kostyamat, 2024-12-22:
Есть мертвый iDrive BMW. Там стоит вот такой датчик оптического энкодера https://www.alldatasheet.com/datasheet-pdf/download/344168/CITIZEN/CPI-250.html 
Судя по схеме, у него общие коллекторы транзисторов. Предполагаю поставить в эммитеры резисторы на массу, сопротивлением скажем 10кОм, на общий коллекторов подать +5в, на всякий случай через резистор 3.2кОм, и снимать сигнал между эмитером и резистором, установив на пинах контроллера PULL_UP.  Когда транзистор закрыт, там должно быть около 1в, что 0, при открытии транзистора должно быть около 3,4 вольта, что точно 1.

Вопрос: будет ли библиотека читать такое, и какой тип энкодера применить в сетапе? И если нет, нельзя ли добавить?

kostyamat, 2024-12-22:
Но там ведь получится логика обратная нормальному энкодеру, то есть вместо 0 будет 1, и вместо 1 будет 0.

GyverLibs, 2024-12-22:
Ну, смотреть графики типов

closed 2024-12-27 by kostyamat