GyverLibs/EncButton #67: EncButtonT использует deprecated метод  bool VirtButton::timeoutState(uint16_t)

Shersula, 2024-12-27:
Использую EncButtonT для обработки right()/left()/rightH()/leftH()/click()
Во время компиляции выдает предупреждение о использовании устаревшего метода timeoutState внутри VirtButton.h

Создание экземпляра энкодера
[image]

Использование энкодера
[image]

Дефайны настроек описаны в platformio.ini
[image]

Вывод во время компиляции:
[image]

GyverLibs, 2024-12-27:
видимо придётся просто убрать deprecated

closed 2024-12-27 by GyverLibs

Shersula, 2024-12-28:
@GyverLibs Это и правда был баг компилятора, не знаю что поменялось, но сегодня без обновления библиотеки и изменения кода, варнинг исчез.
Видимо это сюрпризы со стороны PlatformIO

GyverLibs, 2024-12-28:
Ну я убрал deprecated. По сути что было - внутри deprecated метода вызывался deprecated метод. Сам верхний метод в библиотеке нигде не вызывается, поэтому с чего компилятор на него триггерится непонятно