GyverLibs/EncButton #69: Событие: держание и поворот

diarus, 2025-01-04:
В версии 2 для левого и правого поворота с одновременным удержанием кнопки энкодера-это два отдельных события. (Они выдавали значения 4 и 3) В версии 3.7.0 эти события исчезли и теперь всё отдельно. Удержание  кнопки, левый поворот, правый поворот-это три разных события.  На версии 2 я использовал switch для обработки этих событий. Теперь это невозможно. Нужно переделывать код на if...else
Я правильно понимаю?

GyverLibs, 2025-01-04:
Да, теперь более удобно. Нужно обработать поворот, и внутри условия можно проверить нажатость кнопки, направление итд

diarus, 2025-01-05:
Но в борьбе за скорость проиграли. Вместо быстрой switch придется
использовать if else.
С уважением Черных Андрей Владимирович.
т.+79206741714
г. Иваново

GyverLibs, 2025-01-05:
Ахахахха, ну это уже шизофрения) ну и вроде никто не мешает вызвать
switch (enc.dir()) {
case 1:
case -1:
}

closed 2025-01-05 by diarus