GyverLibs/EncButton #72: Не наступает событие Fast

CodePower50, 2025-05-04:
Гайвер, что должно произойти во всей Вселенной, чтоб наступило событие fast? Запускаю самый быстрый пример из скачанной библиотеки (последняя версия 3.7.2) - isr на прерываниях, как самый быстрый. Ничего не добавлено и ничего не вырезано. Все работает (во всех других примерах так-же), кроме fast. Событие не наступает, как бы быстро не вращался энкодер, то есть - в выводимой в порт информации fast всегда равен 0. Просимулировал пример  в Протеусе - чтоб наступило событие быстрого вращения (fast=1) нужно чтоб у энкодера было где-то 60 импульсов на оборот и частота импульсов с него должна быть больше 50Гц. Вот тогда наступает событие fast, и то не всегда. Контроллер - обычный Ардуино Нано с Атмегой328.

[image]

GyverLibs, 2025-05-04:
> что должно произойти во всей Вселенной

между тиками должно пройти меньше времени, чем задано в setFastTimeout (умолч. 30). Фича очень стабильно работает в железе на всех встроенных примерах

closed 2025-05-04 by GyverLibs

CodePower50, 2025-05-04:
Ну, вот скрин из АрдуиноИДЕ - работа  "в железе". Да, сам по себе пример работает, но только если ничего не трогать. Но если сделать имитацию программы не 100 мс, как в примере ( delay(100) ), а хотя-бы 150 мс то уже начнутся пропуски, и если этот delay увеличивать - то все становится только хуже и хуже, и  примерно от  delay(300) fast совсем перестает работать.

[image]

GyverLibs, 2025-05-04:
В нормальной программе и не должно быть таких диких задержек. Любом случае всё замечательно работает в железе и при 200, и при 300 мс задержки в луп, проверяйте энк/подключение/подтяжки итд